Xilinx/Vitis-AI #826: Problem about videos while running code.

nrllhclb, 2022-06-01:
Under the VART/segmentation path,
i see segmented video frames when i run built-in video example, however in any custom video, video hasnt been segemented, it is only run original video frames.

I have just started to learn deep learning anad Vitis-AI, 
I have two question, first, is there different type video such as gray scale like an image? If yes, which type video is used for segmentation in this repeository?

nrllhclb, 2022-06-02:
I found that it only works specific video type such as raw and webm, however i used an video with extension .webm to run, but it didnt work.

qianglin-xlnx, 2022-08-11:
Hi @nrllhclb 
Is this issue solved?

nrllhclb, 2022-08-12:
It hasn't been solved, also i haven't focused on it for long time. Iwill closed this issue.

closed 2022-08-12 by nrllhclb