Xilinx/Vitis-AI #828: Pytorch quantize error: cannot import name 'Graph'

CharlesChen1024, 2022-06-02:
Hi,
When I quantize the model, i got one error message.
Error message:
Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-optimizer_pytorch/lib/python3.6/site-packages/nndct_shared/compile/xgraph.py", line 20, in <module>
    from xir import Graph
ImportError: cannot import name 'Graph'

I try to use "pip install xir"  but I still got same error. 
How to solve this problem?
Thanks

niuxjxlnx, 2022-06-13:
Hi, @CharlesChen1024:

Which version of VAI are you using?

niuxjxlnx, 2022-06-13:
Hi, @CharlesChen1024 :

I notice you are using optimizer conda env:
/opt/vitis_ai/conda/envs/**vitis-ai-optimizer_pytorch**.

Please do quantization in conda env **vitis-ai-pytorch**.

CharlesChen1024, 2022-06-13:
Hi, @niuxjxlnx : 
Thanks for your reply. I successfully run the same code in conda env vitis-ai-pytorch.

closed 2022-06-13 by CharlesChen1024