Xilinx/Vitis-AI #829: vai_q_pytorch undefined symbol: _ZNK2at6Tensor8toStringEv

JaviMota, 2022-06-03:
Hi

I have installed vitis AI GPU using the docker_build_gpu.sh file and I needed to add this commands to the vitis-ai-gpu.Dockerfile in order to make it work:
RUN rm /etc/apt/sources.list.d/cuda.list
RUN rm /etc/apt/sources.list.d/nvidia-ml.list
RUN apt-key del 7fa2af80
RUN apt-key adv --fetch-keys https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64/3bf863cc.pub
RUN apt-key adv --fetch-keys https://developer.download.nvidia.com/compute/machine-learning/repos/ubuntu1804/x86_64/7fa2af80.pub

Without adding this commands it complains about Nvidia keys. Im using A4000 GPU.

The problem is when I run "from pytorch_nndct.apis import torch_quantizer" in python, it complains about "undefined symbol: _ZNK2at6Tensor8toStringEv"

Do you know how to solve this?

Thank you
Javi

yuchen2580, 2022-06-04:
Same issue. And I believe is also mentioned in #739 .

I thought it might be a pytorch torchvision mismatch issue since similar message might show when torchvision and torch is not competible. After building the container, torch==1.7.1 torchvision==0.8.0 instead of torchvision==0.8.2 as listed in pytorch installation guide.
  
So I try creating another conda env by using 'replace_pytorch.sh'. but it gives me below error:
```

Looking for: ['pytorch==1.8.1', 'torchvision==0.9.1', 'torchaudio==0.8.1', 'cudatoolkit=10.2']

Pinned packages:
  - python 3.6.*

Encountered problems while solving:
  - nothing provides mkl 2019.1.* needed by libblas-3.8.0-4_mkl
  - package pytorch-1.8.1-py3.6_cpu_0 requires numpy >=1.19, but none of the providers can be installed

pytorch/linux-64          
pytorch/noarch            

#### Pytorch packages is NOT replaced correctly.
```

niuxjxlnx, 2022-08-05:
@JaviMota：

Can you check your GPU driver version by tool "nvidia-smi"?

niuxjxlnx, 2022-08-11:
@JaviMota :

Please try vitis-ai 2.5 docker and change pytorch version with the updated script https://github.com/Xilinx/Vitis-AI/blob/master/docker/dockerfiles/replace_pytorch.sh

closed 2022-08-31 by niuxjxlnx