Xilinx/Vitis-AI #830: help with error when running VITIS AI example on U200

dylangeon, 2022-06-03:
Hello there,

When I run this:

Vitis-AI /workspace/examples/DPUCADF8H/tf_inception_v1 > ./inception_example ./inception_v1_tf.xmodel sample_classification.jpg

 

I got error message as follows:

WARNING: Logging before InitGoogleLogging() is written to STDERR

I0527 10:51:45.357604  213 [main.cc](https://main.cc/):293] create running for subgraph: subgraph_InceptionV1/InceptionV1/Conv2d_1a_7x7/Conv2D

terminate called after throwing an instance of 'std::runtime_error'

 what(): Error: Could not acquire CU

Aborted (core dumped)

 

Thanks a lot for any help.

SuZhan0322, 2022-07-19:
hi @dylangeon, 
1. You can load xclbin to device through the "/opt/xilinx/xrt/bin/xbutil program" command.
2. If load xclbin successfully , you can compare whether the versions of xmodel and xclbin match.

SuZhan0322, 2022-08-07:
Hi @dylangeon ,
Is there any update to this issue? Thanks~

qianglin-xlnx, 2023-01-17:
Closing since no activity for more than 4 months, please reopen if you still have any questions, thanks

closed 2023-01-17 by qianglin-xlnx

guoquankisscall, 2023-05-09:
hello, how can i compare whether the versions of xmodel and xclbin match?

guoquankisscall, 2023-05-09:
I have this error "Error: Could not acquire CU" on U200, can you tell me how to check，thank you！