Xilinx/Vitis-AI #833: vai_c_tensorflow2 编译一个具有两个输入，自定义层也有两个输入的模型

hito0512, 2022-06-08:
用 vai_c_tensorflow2 编译一个具有两个输入的模型，指令如下：
`vai_c_tensorflow2 -m ./quant/quantized.h5 -a zcu104_arch.json -o xmodel -n custom_op --options '{"input_shape": {"z": "1,127,127,3", "x": "1,303,303,3"}}' `
其中一个自定义层也有两个输入，报错如下：
`[INFO] Namespace(batchsize=1, inputs_shape=None, layout='NHWC', model_files=['./quant/quantized.h5'], model_type='tensorflow2', named_inputs_shape={'z': '1,127,127,3', 'x': '1,303,303,3'}, out_filename='/tmp/siamFC_custom_op_org.xmodel', proto=None)
in_shapes: {'z': [1, 127, 127, 3], 'x': [1, 303, 303, 3]}
AssertionError: [ERROR] keras CustomOpWrapper requires one input: actual:2`
请问这个如何处理，自定义层含有两个输入的话不能编译吗？
模型结构如下：
[image]

zhenzhen-Xilinx, 2022-06-08:
你好 @hito0512 ，
可以将浮点模型和量化模型发送过来吗？

hito0512, 2022-06-08:
> 你好 @hito0512 ， 可以将浮点模型和量化模型发送过来吗？

模型有点大，有邮箱吗？

zhenzhen-Xilinx, 2022-06-08:
我的邮箱： zhenzhen@xilinx.com

hito0512, 2022-06-10:
> 我的邮箱： [zhenzhen@xilinx.com](mailto:zhenzhen@xilinx.com)

您好，能看出有啥问题吗？是不是我哪里弄错了？

zhenzhen-Xilinx, 2022-06-10:
模型发送过来了吗？我一直没收到相关邮件。

hito0512, 2022-06-10:
啊？当天就发了呀

| |
杨桃
|
|
***@***.***
|
---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2022年6月10日 10:43 |
| 收件人 | ***@***.***> |
| 抄送人 | ***@***.***> ,
***@***.***> |
| 主题 | Re: [Xilinx/Vitis-AI] vai_c_tensorflow2 编译一个具有两个输入，自定义层也有两个输入的模型 (Issue #833) |

模型发送过来了吗？我一直没收到相关邮件。

hito0512, 2022-06-10:
已重新发送，请查收

| |
杨桃
|
|
***@***.***
|
---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2022年6月10日 11:02 |
| 收件人 | ***@***.***> |
| 抄送人 | ***@***.***> ,
Your ***@***.***> |
| 主题 | Re: [Xilinx/Vitis-AI] vai_c_tensorflow2 编译一个具有两个输入，自定义层也有两个输入的模型 (Issue #833) |

啊？当天就发了呀

| |
杨桃
|
|
***@***.***
|
---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2022年6月10日 10:43 |
| 收件人 | ***@***.***> |
| 抄送人 | ***@***.***> ,
***@***.***> |
| 主题 | Re: [Xilinx/Vitis-AI] vai_c_tensorflow2 编译一个具有两个输入，自定义层也有两个输入的模型 (Issue #833) |

模型发送过来了吗？我一直没收到相关邮件。

hito0512, 2022-06-10:
> 我目前还是没有收到邮件。可以告知一下您的邮箱吗？ 我给您发送一封邮件，您回复我这封邮件，同时附带上浮点模型和量化模型。可以吗?

好的，邮箱： hito0512@163.com

zhenzhen-Xilinx, 2022-06-10:
> > 我目前还是没有收到邮件。可以告知一下您的邮箱吗？ 我给您发送一封邮件，您回复我这封邮件，同时附带上浮点模型和量化模型。可以吗?
> 
> 好的，邮箱： [hito0512@163.com](mailto:hito0512@163.com)

模型已经收到，谢谢。如果有进展，我会尽快告知。

hito0512, 2022-06-14:
您好，发现啥问题了吗？ 

| |
杨桃
|
|
***@***.***
|
---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2022年6月10日 14:03 |
| 收件人 | ***@***.***> |
| 抄送人 | ***@***.***> ,
***@***.***> |
| 主题 | Re: [Xilinx/Vitis-AI] vai_c_tensorflow2 编译一个具有两个输入，自定义层也有两个输入的模型 (Issue #833) |

我目前还是没有收到邮件。可以告知一下您的邮箱吗？ 我给您发送一封邮件，您回复我这封邮件，同时附带上浮点模型和量化模型。可以吗?

好的，邮箱： ***@***.***

模型已经收到，谢谢。如果有进展，我会尽快告知。

zhenzhen-Xilinx, 2022-06-15:
> 您好，发现啥问题了吗？

你好，需要确认下‘zcu104_arch.json  ZCU104.dcf’ 这两个文件 是从哪里获取的？

hito0512, 2022-06-15:
这两个文件就是为了zcu104自己做的的，也可以用官方的文件进本编译；  /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU102/arch.json
vai_c_tensorflow2 -m ./quantized/quantized.h5 -a /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU102/arch.json -o ./ -n tf2_custom_op
，那用官方文件替换了吧
| |
杨桃
|
|
***@***.***
|
---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2022年6月15日 09:35 |
| 收件人 | ***@***.***> |
| 抄送人 | ***@***.***> ,
***@***.***> |
| 主题 | Re: [Xilinx/Vitis-AI] vai_c_tensorflow2 编译一个具有两个输入，自定义层也有两个输入的模型 (Issue #833) |

您好，发现啥问题了吗？

你好，需要确认下‘zcu104_arch.json ZCU104.dcf’ 这两个文件 是从哪里获取的？

zhenzhen-Xilinx, 2022-06-15:
好的，请稍等。

zhenzhen-Xilinx, 2022-06-15:
你好，请使用 docker  Vitis AI 2.5 Release 版吧。问题在这个版本中修复了。

hito0512, 2022-06-15:
这边做了测试是可以的吗？

Message ID: ***@***.***>

zhenzhen-Xilinx, 2022-06-15:
已经测试了，可以的。
vai_c_tensorflow2 -m ./quantized.h5 -a /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU102/arch.json -o ./ -n tf2_custom_op

另外量化模型的输入数据和验证数据格式是一样的。

closed 2022-06-17 by hito0512