Xilinx/Vitis-AI #834: Deisgn custom conv2d operation in hardware side

WhiteCri, 2022-06-09:
Hello,
I want to replace some operations(mainly convolution2d) at hardware level.
For example, design some operation with vivado -> create bitstream -> create python/c++ api which call custom operation 
Acutally I don't have enough experience at hardware design, so can you give me some tips?

Thanks in advance,

WhiteCri, 2022-07-15:
I prefer to design new IP, but actually does not care both. 
There exist many proposals of CNN(but not limited only to CNN) design paper, and I'd like to apply them along with Vitis-AI, since it's awesome framework!
Is it possible to apply custom CNN design in anyway?

qianglin-xlnx, 2022-09-06:
Hi @WhiteCri 
You can refer to https://github.com/Xilinx/Vitis-AI/tree/master/examples/Whole-App-Acceleration to know how to make custom IP work with Vitis-AI DPU IP.
But for custom CNN IP, it maybe more complicated. We have to check if the custom cnn op is supported by DPU. If yes, it will be challenge for Xcompiler to compile the model and assign the supported cnn op to CPU side. If no,  you can just design custom cnn ip as a custom ip and apply it along with Vitis-AI, just like: https://github.com/Xilinx/Vitis-AI/tree/master/examples/Whole-App-Acceleration

qianglin-xlnx, 2022-09-26:
Hi @WhiteCri 
Closing since no activity for more than 2 weeks, please reopen if you still have question, thanks

closed 2022-09-26 by qianglin-xlnx