Xilinx/Vitis-AI #837: Vitis-AI Quantizer - KeyError when convert from Pytorch to xmodel

haipnh, 2022-06-10:
Hi,

I am trying to quantize the FADNet. I can run the demo, evaluation, training, and quantization.

But when I converted it to xmodel, it show following error:
```
[VAIQ_NOTE]: =>Converting to xmodel ...
Traceback (most recent call last):
  File "main_qat.py", line 144, in <module>
    main(opt)
  File "main_qat.py", line 55, in main
    best_EPE = trainer.validate()
  File "/workspace/models/AI-Model-Zoo/pt_fadnet_sceneflow_576_960_441G_2.0/code/dltrainer_qat.py", line 341, in validate
    dump_xmodel(self.quant_dir, deploy_check=True)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/apis.py", line 137, in dump_xmodel
    qp.dump_xmodel(output_dir, deploy_check, lstm_app)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 205, in dump_xmodel
    xmodel_depoly_infos, dump_deploy_infos = compiler.get_xmodel_and_dump_infos(quantizer, deploy_graphs)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/compile/xir_compiler.py", line 45, in get_xmodel_and_dump_infos
    graph_quant_info = XirCompiler.get_deloy_graph_infos(quantizer, deploy_graphs_list[0])
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/compile/xir_compiler.py", line 69, in get_deloy_graph_infos
    if len(quant_groups[blob_name]) == 1:
KeyError: 'FADNet::FADNet/DispNetC[dispnetc]/ResBlock[conv3]/ReLU[relu2]/529'
```

niuxjxlnx, 2022-06-13:
@haipnh :

Which version of VAI are you using, and can you share your steps to run the traning, quantization and converting xmodel?

haipnh, 2022-06-14:
My environment:

- AMD Ryzen 5 5600X | RTX 3080 Ti
- Ubuntu 18.04.6 | Kernel 5.4.0-117-generic
- NVIDIA Driver 470 | CUDA 11.4
- Docker version 20.10.16
- Vitis-AI v2.0 commit 996459fb96

After I downloaded the GPU FADNet, it was extracted to Vitis-AI/models/AI-Model-Zoo. 

- I set `devices=0` in config files in `exp_configs` and `run_*.sh`. 
- I set `CUDA_VISIBLE_DEVICES=0` in any `run_*.sh`.

Then I modified the `docker_run.sh` to make a mount point to my dataset directory, and executed it.
`-v /media/HDD_driver/dataset/SceneFlow/Extracted:/workspace/models/AI-Model-Zoo/pt_fadnet_sceneflow_576_960_441G_2.0/data`

Here is my next procedure:
```shell
conda activate vitis-ai-pytorch
pip install --user scikit-image OpenEXR==1.3.2
pip install --user torch==1.7.0+cu110 torchvision==0.8.1+cu110 torchaudio===0.7.0 -f https://download.pytorch.org/whl/torch_stable.html
cd models/AI-Model-Zoo/pt_fadnet_sceneflow_576_960_441G_2.0
bash run_demo.sh
bash run_eval.sh
bash run_train.sh # I stopped after 3 epochs and decided to quantize pre-trained model
bash run_qat.sh # error in this step
```

Daniel-bboy, 2022-06-17:
Have you solved your problem? I have met the same problem with you

yuchen2580, 2022-07-29:
@haipnh @Daniel-bboy 
This seems to be a common error when nndct version is not matched with the one that export the deployable model.
The run_qat.sh will auto qat and generate the deployable model to ./models/fadnet-qat-sceneflow,  the run_deploy.sh reads deployable model under ./qunatized folder by default, have you tried to set the  "model" argument in run_deply.sh line to "./models/fadnet-qat-sceneflow/model_best_deployable.pth"?

Hamptonjc, 2022-08-23:
I've been experiencing this problem as well. QAT works fine in VAI 2.5, but I am constrained to 2.0 DPU thus I need to stay with 2.0. I tried training a model from scratch in the VAI Docker container (to ensure the model is with the same versions), but I still get this KeyError. Any ideas?

yuchen2580, 2022-08-24:
@Hamptonjc 
Hi thanks for your feedback.
I cannot reproduce this problem with Vitis 2.0, docker.

Here is my procedure to retain the network for reference.
1. run the "run_train.sh"  
2. set the "model" variable in "exp_configs/fadnet_qat.conf" to the model generated in step 1 (default path is models/fadnet-sceneflow/model_best_single.pth) and run_qat.sh (you may have to change model_best.pth to model_best_single.pth in the default path)
3. Change the "model" variable in "run_deploy.sh" to point at the deployable model generated from step 2. (default: models/fadnet-sceneflow-qat/model_best_deployable.pth). And run the "run_deploy.sh" t

I suggest you to do a quick test with 1 iteration of training.
Please let me know if this works for you, and also share the error message&commands if possible.

Hamptonjc, 2022-08-25:
@yuchen2580 

I am using a custom script (i.e. not the .sh files). My problem was I was performing training and exporting in one run of my .py script. Once I switched to performing one run for training and one run for exporting, I don't get the key error.

Not sure what's going on, but its working.

closed 2022-09-05 by niuxjxlnx