Xilinx/Vitis-AI #838: Screen flickering while running .webm examples

Werterror, 2022-06-10:
Hi all,
I tried couple examples with display connected directly to ZCU102 via DP.
I launched them via petalinux terminal.
Examples that use photos not videos work just fine, but segmentation and adas_detection make screen flicker.
I made video so you guys could have better idea what I mean:
https://www.youtube.com/watch?v=-2amq1AtFD0&ab_channel=KamilSawicki
Is that a known issue?

TiL497, 2022-06-10:
Hi,
I had the same problem with my 1440p monitor. Its a Known Issue https://github.com/Xilinx/Vitis-AI/blob/master/dsa/DPU-TRD/prj/Vitis/README.md
Lowering the resolution with xrandr fixed the problem in my case.

Werterror, 2022-06-13:
Hi @TiL497,
You were right, I changed resolution to full HD and it works just fine.
Thank you!

closed 2022-06-13 by Werterror