Xilinx/Vitis-AI #839: Problem deploying tutorial on VCK5000

jrobador, 2022-06-10:
Hi everyone, I tried to run the follow tutorial on my VCK5000: https://github.com/Xilinx/Vitis-AI-Tutorials/tree/1.4/Design_Tutorials/12-Alveo-U250-TF2-Classification. 
When I builded the project, I changed the arch.json (because this tutorial is based on U250) and the the options from:
		  --options '{"input_shape": "4,100,100,3"}' to  --options '{"input_shape": "1,100,100,3"}'
Because otherwise I got a lot of warnings. 
After that, when I try to run the deployment, I got segmentation fault (core dumped). I thought it is because I changed the input_shape but I don't know how to fix this problem.
Can anyone help me?

wanghy-xlnx, 2022-06-13:
Hi @jrobador ,

1. First of all, you need to confirm whether the card you are using is VCK5000-ES1 or VCK5000-PROD. In case of VCK5000-ES1, Vitis AI 1.4 or 1.4.1 can be used. In case of VCK5000-PROD, you need to use Vitis AI 2.0 on the master branch or or later updated version.
2. You need to make sure that the docker image, platform, XRT version, etc. used correspond to the same Vitis AI version. Refer to [Docker image version](https://github.com/Xilinx/Vitis-AI/blob/master/docs/quick-start/install/install_docker/load_run_docker.md), Refer to [vck5000 platform install](https://github.com/Xilinx/Vitis-AI/tree/master/setup/vck5000).
3. You can first run the [VART samples](https://github.com/Xilinx/Vitis-AI/blob/master/demo/VART/README.md) or the [Vitis AI Library samples](https://github.com/Xilinx/Vitis-AI/blob/master/demo/Vitis-AI-Library/README.md) to make sure the environment is correct.

SuZhan0322, 2022-08-07:
Hi @jrobador ,
Is there any update to this issue?  
Another question that needs to be clarified is that there are any warnings in the compiling process？I'm just worried that this model has ops that vck5000-es1 doesn't support. And This is a possibility to cause segmentation fault.

jrobador, 2022-08-07:
Hi @SuZhan0322, when I had --options '{"input_shape": "4,100,100,3"}' at compile.sh, I got [this warning output](https://github.com/Xilinx/Vitis-AI/files/9277330/output_1.txt). Using this model, I had the following [deploy output](https://github.com/Xilinx/Vitis-AI/files/9277333/deployoutput.txt). 
For this reason, I change the --options '{"input_shape": "4,100,100,3"}' to --options '{"input_shape": "1,100,100,3"}'. With this options I didn't have error compiling. Then, I run the deploy command and have the following ouput message
[output message](https://github.com/Xilinx/Vitis-AI/files/9277339/deployoutput2.txt).

If you want, you can contact me at my email to keep working on this issue.
 joaquinrobador@gmail.com