Xilinx/Vitis-AI #841: DPU_Tensor Input shape Isuue

AfifaIshtiaq, 2022-06-14:
Hi, 
Can you tell me if my input dimension is 10,14,72,2  why input tenosr only shows 5,14,72,2

def init_dpu_runner(dpu_runner):
  '''
  Setup DPU runner in/out buffers and dictionary
  '''

  io_dict = {}
  inbuffer = []
  outbuffer = []

  # create input buffer, one member for each DPU runner input
  # add inputs to dictionary
  dpu_inputs = dpu_runner.get_input_tensors()
  print('dpu_inputs',dpu_inputs)
  i=0
  for dpu_input in dpu_inputs:
    print('DPU runner input:',dpu_input.name,' Shape:',dpu_input.dims)
    inbuffer.append(np.empty(dpu_input.dims, dtype=np.float32, order="C"))
    io_dict[dpu_input.name] = inbuffer[i]
    i += 1

  # create output buffer, one member for each DPU runner output
  # add outputs to dictionary
  dpu_outputs = dpu_runner.get_output_tensors()
  i=0
  for dpu_output in dpu_outputs:
    print('DPU runner output:',dpu_output.name,' Shape:',dpu_output.dims)
    outbuffer.append(np.empty(dpu_output.dims, dtype=np.float32, order="C"))
    io_dict[dpu_output.name] = outbuffer[i]
    i += 1

  return io_dict, inbuffer, outbuffer

rx_data_input Shape: (108, 10, 14, 72, 2)
dpu_inputs [<xir.Tensor named 'quant_RX-Data-In'>, <xir.Tensor named 'quant_TX-Pilot-In'>, <xir.Tensor named 'quant_Raw-Channel-Est-In'>]
DPU runner input: quant_RX-Data-In  Shape: [5, 14, 72, 2]
DPU runner input: quant_TX-Pilot-In  Shape: [5, 14, 72, 2]
DPU runner input: quant_Raw-Channel-Est-In  Shape: [5, 14, 72, 2]
DPU runner output: quant_Conv-Out_relu_fix  Shape: [5, 14, 72, 2]

zz002, 2022-06-23:
Hi @AfifaIshtiaq 
Currently our library only supports four dimensions [batch, h, w, c], where batch is related to your board. If you use vck190 of c32b5 then batch=5, you can fuse 10 in [10,14,72,2] to other dimensions, such as [1,14,72,20] when training the model. Then you will get the input is [5, 14, 72, 20], where 5 means that the model can process 5 sets of [1, 14, 72, 20] data in parallel

AfifaIshtiaq, 2022-06-23:
Thanks alot it was very helpful

closed 2022-07-13 by qianglin-xlnx