Xilinx/Vitis-AI #842: VAI trace Vitis AI analyzer Issue

AfifaIshtiaq, 2022-06-14:
Why the vai trace output shows a summary for only one subgraph  on the vitis ai analyzer while on svg I have multiple dpu subgraphs
[image]

[image]

li900309, 2022-06-15:
the result you got is for the root su-bgraph (a,k.a top-level sub-graph) if you want more detailed profiling result, you can try run vaitrace with "--fine-grained"
[image]

AfifaIshtiaq, 2022-06-15:
> the result you got is for the root sub-graph (a,k.a top-level sub-graph) if you want more detailed profiling result, you can try run vaitrace with "--fine-grained" option https://github.com/Xilinx/Vitis-AI/tree/master/examples/Vitis-AI-Profiler#command-line-usage [image]

I was able to produce fine grained results. Can you please tell me what is meant by runs? Also in fine grained I'm not able to see Vitis AI performance tab anymore how can I get it
[image]

li900309, 2022-06-24:
runs = the number of times this sub-graph is called
about the FPS view, I think in --fine_grained mode the CNN model will be spitted into several subgraphs and executed one-by-one, in this situation the FPS number become meaningless.

qianglin-xlnx, 2022-08-11:
Hi @AfifaIshtiaq 
Is this issue solved?

qianglin-xlnx, 2022-08-31:
Closing since no activity for more than 2 weeks, please reopen if you still have question, thanks

closed 2022-08-31 by qianglin-xlnx

gunertatar, 2022-11-27:
Hi;
I m using Kria KV260 vision board and i executed the tf_resne50 example from the Vitis AI VART. I d like to look at the performance of behavior of the board and used the vaitrace ./resnet50 resnet50.xmodel. After execution i got only " profile_summry.csv, summary.csv, vart_trace.csv, vitis_ai_profile.summary, xrt.run_summary" files. However, when i look around i think, xclbin.ex.run_summary file also should be in the executed file. By the way Vitis unifed software platform and vitis_analyzer version 2021.1.
Is there anyone could help me please? I really need to check all files belongs to execution such as HAL_API Calls, OpenCL data, Data Transfer, DPU, buffer exchange and so on.

Thanks

anirudhitagi, 2023-01-04:
@gunertatar , I am also facing a similar issue. I am working with ZCU104 and I did the python flow:  `python3 -m vaitrace_py ./resnet50.py 2 /usr/share/vitis_ai_library/models/resnet50/resnet50.xmodel`
It generated the same files that you got. I tried importing the xrt.run_summary on vitis_analyzer, but the graphs and results it shows are missing lot of information.
Have you found any solution?
 
 @qianglin-xlnx  any advice?

gunertatar, 2023-01-04:
> @anirudhitagi Unfortunately i haven't found any solution yet. And no one has given information about that. If you find please let me know, or vice-versa i ll let you know.

li900309, 2023-01-06:
hi @anirudhitagi @gunertatar  the file name of result was changed since vitis and xrt 2021.1 version, xclbin.ex.run_summary was replaced by xrt.run_summary, so just import xrt.run_summary to vitis_analyzer.
for more details, you can try vaitrae --fine_grained , and also you can ntry vaitrace --txt which will show a text summary

anirudhitagi, 2023-01-06:
I see. Thank you for your response.

dextroza, 2023-02-09:
Hi,
I am getting error when trying to run ai profiler on VART 2.5, petalinux 22.1. I am trying to profile my c++ application on ZCU104.
The code:
`$ vaitrace ./app_zcu104.elf`,
The error occurs at the end of profiling:
`APM Stop Collecting
WARNING:root:Cannot open 'zyxclmm_drm/ip_layout'
INFO:root:Generating VTF
INFO:root:Processing xmodel information
INFO:root:SubGraph index duplication: [subgraph_data_upload_0|0]
Traceback (most recent call last):
  File "/usr/bin/vaitrace", line 186, in <module>
    main(*parseCmdLine())
  File "/usr/bin/vaitrace", line 181, in main
    vaitraceWriter.write(options)
  File "/usr/bin/xlnx/vaitrace/vaitraceWriter.py", line 83, in write
    writer.vtf.convert.xat_to_vtf(rawData)
  File "/usr/bin/xlnx/vaitrace/writer/vtf/convert.py", line 331, in xat_to_vtf
    convert_dpu(xat.get('vart'))
  File "/usr/bin/xlnx/vaitrace/writer/vtf/convert.py", line 140, in convert_dpu
    dpu_profile_summary = dpu_parser.get_dpu_profile_summary()
  File "/usr/bin/xlnx/vaitrace/writer/parser/vart_parser.py", line 334, in get_dpu_profile_summary
    perf = workload * batch / avg_t * 1000  # to GOP/ms
ZeroDivisionError: float division by zero`

How to fix this error? Thanks!