Xilinx/Vitis-AI #853: The release file of  VAI 2.5 is still for VAI 2.0

hill19850213, 2022-06-15:
I unzipped VAI 2.5 file and found the contents were still for vai 2.0. Is the file incorrect?

dsp-avt, 2022-06-15:
@hill19850213 - you're talking about the Vitis-AI/docker/dockerfiles/VERSION.txt file correct?

The file has 2.0.0.1103, but I think it should actually be 2.5.0.1260

hill19850213, 2022-06-15:
Here is screenshot for unzipped file of Vitis-AI-2.5.zip
[image]
This contents are same as [Vitis-AI-2.0](https://github.com/Xilinx/Vitis-AI/tree/996459fb96cb077ed2f7e789d515893b1cccbc95). Actually, the contents are should be same as  [Vitis-AI-2.5](https://github.com/Xilinx/Vitis-AI)

qianglin-xlnx, 2022-06-23:
Hi @hill19850213 
Sorry about the late reply. It's fixed. Could you please try it again, thank you.

closed 2022-06-24 by hill19850213