Xilinx/Vitis-AI #854: Vitis AI compiler: *** Check failure stack trace: ***

AfifaIshtiaq, 2022-06-15:
Why it aborts? I have attached the model 
vitis-ai-tensorflow2) Vitis-AI /workspace/models/ChEstModel > vai_c_tensorflow2 -m  quantized_dilationrate_fullytrained.h5 -a /opt/vitis_ai/compiler/arch/DPUCVDX8G/VCK190/VCK190arch.json -o compile -n deeprx_compiledilationFullTrain
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(batchsize=1, inputs_shape=None, layout='NHWC', model_files=['quantized_dilationrate_fullytrained.h5'], model_type='tensorflow2', named_inputs_shape=None, out_filename='/tmp/deeprx_compiledilationFullTrain_org.xmodel', proto=None)
[INFO] tensorflow2 model: /workspace/models/ChEstModel/quantized_dilationrate_fullytrained.h5
[INFO] keras version: 2.6.0
[INFO] Tensorflow Keras model type: functional
[INFO] parse raw model     :100%|███████████████████████████████████████████████████████████████████| 103/103 [00:00<00:00, 18023.84it/s]             
[INFO] infer shape (NHWC)  :100%|███████████████████████████████████████████████████████████████████| 176/176 [00:00<00:00, 2991.78it/s]              
[INFO] perform level-0 opt :100%|███████████████████████████████████████████████████████████████████| 2/2 [00:00<00:00, 63.27it/s]                    
[INFO] perform level-1 opt :100%|███████████████████████████████████████████████████████████████████| 2/2 [00:00<00:00, 245.36it/s]                   
[INFO] generate xmodel     :100%|███████████████████████████████████████████████████████████████████| 176/176 [00:00<00:00, 3657.77it/s]              
[INFO] dump xmodel: /tmp/deeprx_compiledilationFullTrain_org.xmodel
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCVDX8G_ISA2_C64B5_0602001036088252
[UNILOG][INFO] Graph name: DeepRx, with op num: 420
[UNILOG][INFO] Begin to compile...
[UNILOG][FATAL][XCOM_ACGEN_ERROR][instuction generating fail, please contact us.] 
*** Check failure stack trace: ***
Aborted (core dumped)
[deep_rx_dilation_rate_fully_trained.zip](https://github.com/Xilinx/Vitis-AI/files/8909242/deep_rx_dilation_rate_fully_trained.zip)

chuo1, 2022-06-20:
> [quantized_dilationrate_fullytrained.zip](https://github.com/Xilinx/Vitis-AI/files/8909288/quantized_dilationrate_fullytrained.zip) Why it aborts? I have attached the model vitis-ai-tensorflow2) Vitis-AI /workspace/models/ChEstModel > vai_c_tensorflow2 -m quantized_dilationrate_fullytrained.h5 -a /opt/vitis_ai/compiler/arch/DPUCVDX8G/VCK190/VCK190arch.json -o compile -n deeprx_compiledilationFullTrain
> 
> * VITIS_AI Compilation - Xilinx Inc.
> 
> [INFO] Namespace(batchsize=1, inputs_shape=None, layout='NHWC', model_files=['quantized_dilationrate_fullytrained.h5'], model_type='tensorflow2', named_inputs_shape=None, out_filename='/tmp/deeprx_compiledilationFullTrain_org.xmodel', proto=None) [INFO] tensorflow2 model: /workspace/models/ChEstModel/quantized_dilationrate_fullytrained.h5 [INFO] keras version: 2.6.0 [INFO] Tensorflow Keras model type: functional [INFO] parse raw model :100%|███████████████████████████████████████████████████████████████████| 103/103 [00:00<00:00, 18023.84it/s] [INFO] infer shape (NHWC) :100%|███████████████████████████████████████████████████████████████████| 176/176 [00:00<00:00, 2991.78it/s] [INFO] perform level-0 opt :100%|███████████████████████████████████████████████████████████████████| 2/2 [00:00<00:00, 63.27it/s] [INFO] perform level-1 opt :100%|███████████████████████████████████████████████████████████████████| 2/2 [00:00<00:00, 245.36it/s] [INFO] generate xmodel :100%|███████████████████████████████████████████████████████████████████| 176/176 [00:00<00:00, 3657.77it/s] [INFO] dump xmodel: /tmp/deeprx_compiledilationFullTrain_org.xmodel [UNILOG][INFO] Compile mode: dpu [UNILOG][INFO] Debug mode: function [UNILOG][INFO] Target architecture: DPUCVDX8G_ISA2_C64B5_0602001036088252 [UNILOG][INFO] Graph name: DeepRx, with op num: 420 [UNILOG][INFO] Begin to compile... [UNILOG][FATAL][XCOM_ACGEN_ERROR][instuction generating fail, please contact us.] *** Check failure stack trace: *** Aborted (core dumped) [deep_rx_dilation_rate_fully_trained.zip](https://github.com/Xilinx/Vitis-AI/files/8909242/deep_rx_dilation_rate_fully_trained.zip)

I'm trying to requantize the floating point model(deep_rx_dilation_rate_fully_trained.zip) you provided, but I'm having a problem loading the model, can you provide the relevant information.
 quantized_model = load_model(float_model_path)
f'Unknown {printable_module_name}: {class_name}. Please ensure this '
ValueError: Unknown optimizer: Addons>LAMB. Please ensure this object is passed to the `custom_objects` argument. See https://www.tensorflow.org/guide/keras/save_and_serialize#registering_the_custom_object for details.

AfifaIshtiaq, 2022-06-20:
quantized_model = load_model(float_model_path, compile=False) On Mon 20.

chuo1, 2022-06-20:
> 
Because I don't have your calib_dataset, I randomly generated the calib_dataset to do the quantization and then compiled it without error. Can you provide calib_dataset to see it's the data problem?
calib_dataset = np.random.rand(1, *quantized_model.input_shape[0][1:]).astype(np.float32)
calib_dataset=(calib_dataset,calib_dataset,calib_dataset)

chuo1, 2022-06-20:
> > > Dataset is big so I shared some of them. I was able to quantize and compile the model with this dataset for model trained with few epochs. Iam compiling with vck190 ES version
> > 
> > 
> > Because I don't have your calib_dataset, I randomly generated the calib_dataset to do the quantization and then compiled it without error. Can you provide calib_dataset to see it's the data problem? calib_dataset = np.random.rand(1, *quantized_model.input_shape[0][1:]).astype(np.float32) calib_dataset=(calib_dataset,calib_dataset,calib_dataset)
> 
> [dataset.zip](https://github.com/Xilinx/Vitis-AI/files/8938990/dataset.zip)

I tried to quantify and compile with the data you provided, and it works fine. I guess it may be that the version of the quantification [tool(vai_q_tensorflow2) is low. Can you try to update the latest version for quantification?

AfifaIshtiaq, 2022-06-20:
How can I do that can you please help me with it? I'm using VAI2.0 recipe

chuo1, 2022-06-20:
> How can I do that can you please help me with it? I'm using VAI2.0 recipe
> […](#)

What is your docker version please? and how to get it?

> How can I do that can you please help me with it? I'm using VAI2.0 recipe
> […](#)

What is your docker version please? and how to get it?

AfifaIshtiaq, 2022-06-20:
> > How can I do that can you please help me with it? I'm using VAI2.0 recipe
> > […](#)
> 
> What is your docker version please? and how to get it?
> 
> > How can I do that can you please help me with it? I'm using VAI2.0 recipe
> > […](#)
> 
> What is your docker version please? and how to get it?

I'm using this https://github.com/Xilinx/Vitis-AI/releases/tag/v2.0

chuo1, 2022-06-20:
Can you try using version 2.5, it's probably ok

> > > How can I do that can you please help me with it? I'm using VAI2.0 recipe
> > > […](#)
> > 
> > 
> > What is your docker version please? and how to get it?
> > > How can I do that can you please help me with it? I'm using VAI2.0 recipe
> > > […](#)
> > 
> > 
> > What is your docker version please? and how to get it?
> 
> I'm using this https://github.com/Xilinx/Vitis-AI/releases/tag/v2.0

Can you try using version 2.5, it's probably ok

AfifaIshtiaq, 2022-06-20:
> Can you try using version 2.5, it's probably ok
> 
> > > > How can I do that can you please help me with it? I'm using VAI2.0 recipe
> > > > […](#)
> > > 
> > > 
> > > What is your docker version please? and how to get it?
> > > > How can I do that can you please help me with it? I'm using VAI2.0 recipe
> > > > […](#)
> > > 
> > > 
> > > What is your docker version please? and how to get it?
> > 
> > 
> > I'm using this https://github.com/Xilinx/Vitis-AI/releases/tag/v2.0
> 
> Can you try using version 2.5, it's probably ok

Yeah I can try that but I was concerned about what's the problem that's leading to this issue while VAI2.5 doesn't have this issue

AfifaIshtiaq, 2022-06-20:
> > > Can you try using version 2.5, it's probably ok
> > > > > > How can I do that can you please help me with it? I'm using VAI2.0 recipe
> > > > > > […](#)
> > > > > 
> > > > > 
> > > > > What is your docker version please? and how to get it?
> > > > > > How can I do that can you please help me with it? I'm using VAI2.0 recipe
> > > > > > […](#)
> > > > > 
> > > > > 
> > > > > What is your docker version please? and how to get it?
> > > > 
> > > > 
> > > > I'm using this https://github.com/Xilinx/Vitis-AI/releases/tag/v2.0
> > > 
> > > 
> > > Can you try using version 2.5, it's probably ok
> > 
> > 
> > Yeah I can try that but I was concerned about what's the problem that's leading to this issue while VAI2.5 doesn't have this issue
> 
> Due to the large value range of fixed-point data during the addition operation, the compilation cannot be performed normally. VAI2.5 adds the correction function of shift read.

Thanks for the info

closed 2022-09-06 by renjiangyong