Xilinx/Vitis-AI #855: recipe-vitis-ai/vart/var

nrllhclb, 2022-06-15:


JH989876525, 2022-06-16:
I came into the similar error too. 
[image]
According to the build.log, it seems theres some issue of the download url.
- http://petalinux.xilinx.com/sswreleases/rel-v2022/downloads/git2_gitenterprise.xilinx.com.Vitis.vitis-ai-staging.tar.gz
```
Other commonly useful commands are:
 - 'devtool' and 'recipetool' handle common recipe tasks
 - 'bitbake-layers' handles common layer tasks
 - 'oe-pkgdata-util' handles common target package tasks
[INFO] Generating workspace directory
NOTE: Starting bitbake server...
NOTE: Started PRServer with DBfile: /media/jhh/2022/BSP/0615/xilinx-kv260-starterkit-2022.1/build/cache/prserv.sqlite3, Address: 127.0.0.1:44239, PID: 31669
INFO: Specified workspace already set up, leaving as-is
NOTE: Started PRServer with DBfile: /media/jhh/2022/BSP/0615/xilinx-kv260-starterkit-2022.1/build/cache/prserv.sqlite3, Address: 127.0.0.1:40381, PID: 31778
NOTE: Resolving any missing task queue dependencies
Sstate summary: Wanted 1342 Local 809 Network 0 Missed 533 Current 3208 (60% match, 88% complete)
NOTE: Executing Tasks
NOTE: Setscene tasks completed
NOTE: Running task 4351 of 11224 (/media/jhh/2022/BSP/0615/xilinx-kv260-starterkit-2022.1/project-spec/meta-user/recipes-vitis-ai/unilog/unilog_2.5.bb:do_fetch)
NOTE: recipe unilog-2.5-r0: task do_fetch: Started
ERROR: unilog-2.5-r0 do_fetch: Network access disabled through BB_NO_NETWORK (or set indirectly due to use of BB_FETCH_PREMIRRORONLY) but access requested with command /usr/bin/env wget -t 2 -T 30 --passive-ftp -P /media/jhh/2022/LMirror2022.1/kv260/downloads 'http://petalinux.xilinx.com/sswreleases/rel-v2022/downloads/git2_gitenterprise.xilinx.com.Vitis.vitis-ai-staging.tar.gz' (for url http://petalinux.xilinx.com/sswreleases/rel-v2022/downloads/git2_gitenterprise.xilinx.com.Vitis.vitis-ai-staging.tar.gz)
ERROR: Logfile of failure stored in: /media/jhh/2022/BSP/0615/xilinx-kv260-starterkit-2022.1/build/tmp/work/cortexa72-cortexa53-xilinx-linux/unilog/2.5-r0/temp/log.do_fetch.405
NOTE: recipe unilog-2.5-r0: task do_fetch: Failed
ERROR: Task (/media/jhh/2022/BSP/0615/xilinx-kv260-starterkit-2022.1/project-spec/meta-user/recipes-vitis-ai/unilog/unilog_2.5.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4560 tasks of which 4559 didn't need to be rerun and 1 failed.
ERROR: Failed to build project. Check the /media/jhh/2022/BSP/0615/xilinx-kv260-starterkit-2022.1/build/build.log file for more details...
```

tianfang-fafafa, 2022-06-16:
hi @nrllhclb ,
I found that your project is v2021.1, but this recipe is for v2022.1 which use new syntax. 
another problem, this recipe which is for v2022.1 need to be updated after vai2.5 release. Please wait a while

tianfang-fafafa, 2022-06-16:
hi @nrllhclb  @JH989876525 ,
Please modify recipes-vitis-ai/vitis-ai-library/vitisai.inc if you want to try as soon as possible:
    BRANCH = "master"
    SRC_URI = "git://github.com/Xilinx/Vitis-AI.git;protocol=https;branch=${BRANCH}"
    SRCREV = "c26eae36f034d5a2f9b2a7bfe816b8c43311a4f8"

nrllhclb, 2022-06-16:
Thank you for ypur answer, @tianfang-fafafa 
But, I modified vitisai.inc file, however I still get same error, by the way I use Vitis 2021.1 peta 2021.1 vivado 2021.1 vitis latest version

Error message is below; (Undermost, the message after modifiying vitisai.inc )
[image]

tianfang-fafafa, 2022-06-16:
hi @nrllhclb , please try recipe of vai2.5 with petalinux v2022.1  which use new syntax not compitable with v2021.1

qianglin-xlnx, 2022-07-12:
Hi @nrllhclb 
Is this issue solved?

nrllhclb, 2022-07-25:
we abandoned this method, we have worked on different issues.

closed 2022-07-25 by nrllhclb