Xilinx/Vitis-AI #857: How to access individual DPU core of VCK5000 for multi video output

tiny-dentist, 2022-06-16:
Hi, As far as I know VCK5000 has a 8 core DPU, is it possible that each particular core can be accessed individually so that separate input and output can be extracted from it using the Python VART API?

wanghy-xlnx, 2022-06-23:
Hi @tiny-dentist ,

For an 8-core DPU, you can set batchSize=8, so that each start of the DPU operation can calculate 8 inputs and get 8 outputs. For example, [prepare multiple inputs](https://github.com/Xilinx/Vitis-AI/blob/master/examples/VART/resnet50/src/main.cc#L204) in a VART sample. You can get the input and output of each specific core by index.

It should be noted that 8 cores can only be started at the same time, and run the same model. It's just that each specific core has its own input and output.

SuZhan0322, 2022-08-07:
Hi @tiny-dentist ,
Is there any update to this issue? thanks

closed 2022-09-01 by SuZhan0322