Xilinx/Vitis-AI #858: annot find or create target with fingerprint=0x0501000000000fee

Qiang-Flute, 2022-06-16:
When I do the command:
```
./test_jpeg_classification resnet50_pt sample_classification.jpg
```
The log is
```
WARNING: Logging before InitGoogleLogging() is written to STDERR
F0616 03:34:07.241921   274 target_factory.cpp:128] [UNILOG][FATAL][TARGET_FACTORY_UNREGISTERED_TARGET][Unregistered target!] Cannot find or create target with fingerprint=0x0501000000000fee
*** Check failure stack trace: ***
Aborted (core dumped)
```

wanghy-xlnx, 2022-06-28:
Hi @Qiang-Flute ,

According to the Log, I see that you are using a VCK5000-ES1 card. For VCK5000-ES1, you need to use the components corresponding to [Vitis AI 1.4.1](https://github.com/Xilinx/Vitis-AI/tree/v1.4.1).

The steps involved are as follows:
1. Install the correct version of XRT, XRM, XCLBIN and Shell package  on the host according to the [README](https://github.com/Xilinx/Vitis-AI/tree/v1.4.1/setup/vck5000)
2. Run docker image vitis-ai-cpu:1.4.1.978 according [Docker Image List](https://github.com/Xilinx/Vitis-AI/blob/v1.4.1/docs/quick-start/install/install_docker/load_run_docker.md). Run `source ./setup.sh` in a docker container according [README](https://github.com/Xilinx/Vitis-AI/tree/v1.4.1/setup/vck5000)
3. Run Vitis-AI-Library samples according [README](https://github.com/Xilinx/Vitis-AI/blob/v1.4.1/demo/Vitis-AI-Library/README.md)
4. Note that the model used also needs to be downloaded from the [model zoo in version 1.4.1](https://github.com/Xilinx/Vitis-AI/tree/v1.4.1/models/AI-Model-Zoo/model-list)

SuZhan0322, 2022-07-19:
This issue is caused by using mismatched docker images. And the finger(0x0501000000000fee) is only supported in vai1.4.1 docker images. After vai1.4.1, vck5000 es1 are no longer released.

SuZhan0322, 2022-08-07:
Hi @Qiang-Flute ,
Is there any update to this issue? Thanks

qianglin-xlnx, 2022-08-23:
Hi @Qiang-Flute 
Closing since no activity for more than 2 weeks, please reopen if you still have question, thanks

closed 2022-08-23 by qianglin-xlnx