Xilinx/Vitis-AI #859: Fail to build docker-gpu for version 2.5

chuong98, 2022-06-16:
Hello, after running the command `docker_build_gpu.sh`, my computer just hang, and does not show any progress for more than 5h. Then I have to cancel. 
[image]

SuHuynh, 2022-06-17:
I also had the same problem when running command `docker_build_gpu.sh`. Anyone has solution?

hanxue, 2022-06-20:
Hi,

There seems to be a problem with the conda channel link when building GPU docker. I am making some changes internally and verifying it, and will update the docker_build_gpu.sh script within the next 24 hours.

hanxue, 2022-06-22:
Good news, the docker_build_gpu.sh script has been verified and updated. Please can you run `git pull` to refresh your local copy of Vitis AI repository and try running `sh docker_build_gpu.sh` script again?

TC01, 2022-06-22:
Thanks for looking into this! I at least was able to build the container successfully.

LorenzoSun-V, 2022-06-23:
> Hi,
> 
> There seems to be a problem with the conda channel link when building GPU docker. I am making some changes internally and verifying it, and will update the docker_build_gpu.sh script within the next 24 hours.

Thanks for your work ! It works well now!

code-junki, 2022-06-23:
@hanxue are you sure the issue for the GPU docker is resolved? I did a pull few minutes ago and same result:

JaviMota, 2022-06-27:
Hi 

I have cloned the repo again and the issue is still there. Same problem than @code-junki.

Thank you.

janifer112x, 2022-07-26:
this issue has been fixed in the latest commit  on master branch

closed 2022-07-26 by janifer112x