Xilinx/Vitis-AI #860: YoloV4 VOC - 3.0) YoLov4 using VOC Data Set and converting to Caffe Tutorial - Not working on the board

vandenplas9a, 2022-06-16:
YoloV4 VOC
----------------
I’m following the instructions 
[Vitis-AI-Tutorials/readme.md at 1.4 · Xilinx/Vitis-AI-Tutorials · GitHub](https://github.com/Xilinx/Vitis-AI-Tutorials/blob/1.4/Design_Tutorials/07-yolov4-tutorial/readme.md)
3.0) YoLov4 using VOC Data Set and converting to Caffe Tutorial

I’ve made all the recommended changes as far as I can see

I can run a tested on an image on the PC, and the results look as expected

For example:
./darknet detect cfg/yolov4voc.cfg backup/yolov4voc_final.weights /VOCdevkit/VOC2007/JPEGImages/004047.jpg
But:
When I try and run it on the board (I’m using ZCU102, with the ZCU102 image (I could try the ZCU104))
I get
I0709 17:05:00.920222  1154 demo.hpp:744] DPU model size=208x208
Before aborting. (No idea where it is getting the 208x208 from?)

jimheaton, 2022-06-16:
What version of Vitis-AI are you using, and are you using the Vitis-AI Linux Image?
Are you using the Vitis-AI-Library C example code to run on the ZCU102?

I've seen issues when the Vitis-AI Linux Image version does not match the Vitis-AI-Library version.

quentonh, 2022-06-16:
@vandenplas9a I don't mean to be picky, but it preferential if you can leverage either the forum for questions or Github, for confirmed issues, versus posting in both places.  Please see the link that I posted on the forum and confirm if it helps you to resolve this.

yongshengxu-xlnx, 2022-06-23:
About where change leaky-relu from 0.1 to 0.101562:
Change your caffe prototxt ,like this:
layer {
    bottom: "layer0-conv"
    top: "layer0-conv"
    name: "layer0-act"
    type: "ReLU"
    relu_param {
        negative_slope: 0.101562
    }
change negative_slope 0.1 to 0.101562 . You need change all negative_slope val in this prototxt file.

vandenplas9a, 2022-06-23:
Thanks - I think that this has solved the issue:

[UNILOG][WARNING] xir::Op{name = layer152-conv, type = conv2d-fix} has been assigned to CPU: [DPU does not support activation type: LEAKYRELU. Its alpha is 0.101562, but DPU only support 0.101562.].

Ecellent!

I've still got the other issue, when it is starting at the wrong input layer

vai_c_caffe 

options "{'input_shape': '1,416,416,3','mode':'normal','save_kernel':''}"

fails - so, it is not picking up the correct input shape.

I0709 17:05:00.920222 1154 demo.hpp:744] DPU model size=208x208

qianglin-xlnx, 2023-01-12:
Closing since no activity for more than half a year, please open a new issue if you still have any questions, thanks.

closed 2023-01-12 by qianglin-xlnx