Xilinx/Vitis-AI #862: Fail create Board Image of KV260 in Vitis AI v2.5

watchara-knot, 2022-06-17:
After download sd card system from here https://www.xilinx.com/member/forms/download/design-license-xef.html?filename=xilinx-kv260-dpu-v2022.1-v2.5.0.img.gz

Use Etcher software can't finish create.

qianglin-xlnx, 2022-06-18:
HI @watchara-knot 
Thank you for your feedback. We will fix it soon

watchara-knot, 2022-06-18:
@qianglin-xlnx Thx you for fast respone

if you already fix, please let me know

Cloud your help to tell how long the new version will update?

qianglin-xlnx, 2022-06-18:
@watchara-knot 
In the next few days.

qianglin-xlnx, 2022-06-21:
Hi @watchara-knot 
The bug is fixed. Could you please download it again?

watchara-knot, 2022-06-21:
@qianglin-xlnx Thx you for the fast response, I will test tonight and let you know.

more this model zoo
https://github.com/Xilinx/Vitis-AI/blob/master/model_zoo/model-list/tf_rcan_DIV2K_360_640_0.98_86.95G_2.5/model.yaml
in part 
- name: tf_rcan_DIV2K_360_640_0.98_86.95G_2.5
  type: float & quantized
  board: GPU
  download link: https://www.xilinx.com/bin/public/openDownload?filename=tf_rcan_DIV2K_360_640_0.98_86.95G_2.5.zip
  checksum: 1e653e1f3cee29c17af29d3c39c59c78
  
No information about how to export to .xmodel, which way I should contact to get information?

watchara-knot, 2022-06-22:
@qianglin-xlnx Thx you, new file works well

you can close this issue by please suggest how to know the information of post before

closed 2022-06-24 by qianglin-xlnx