Xilinx/Vitis-AI #866: A model verification tool ?

mehdi-nait, 2022-06-19:
It would be really useful if there's a verification tool for custom models, to show unsupported operations and layers that could potentially crash the quantization process. I've been using a custom pytorch YoloV4 model with VAI, I spent a lot of time debugging the quantization with error messages that were little to no help.

qianglin-xlnx, 2022-06-22:
Hi @mehdi-nait 
Thank you for your advice.  
@andyluo7

qianglin-xlnx, 2022-06-22:
HI @mehdi-nait 
In VAI2.5 Model Inspector is introduced. It can inspects a float model and shows partition results
[image]

For tensorflow2 model, you can refer to: https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Inspecting-the-Float-Model
For Pytorch model, you can refer to: https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Inspect-Float-Model-Before-Quantization

closed 2022-07-13 by qianglin-xlnx