Xilinx/Vitis-AI #867: Error about large model quantization

jzymessi, 2022-06-20:
When I quantified a 32GB model, I encountered the following error message
[image]
Is the model too big for the maximum limit?

niuxjxlnx, 2022-07-04:
@jzymessi ：

As the message shows, protobuf, which is dependency of XIR, does not support graph larger than 2GB. 
You can split the graph to some smaller sub-graphs by partial quantization OP(quantStub/dequantStub pairs).

jzymessi, 2022-07-05:
@niuxjxlnx :
Thanks for your reply, by the way,If xmodel size is larger than VCK5000 ddr capacity，xmodel can be splited into multiple xmodel at compile time?
Thanks.

wanghong4compiler, 2022-07-28:
Hi @jzymessi ,
If the required DDR size is larger than hardware's, compiler would report fail and abort, it can't split the model into multiple parts.

closed 2022-08-03 by niuxjxlnx