Xilinx/Vitis-AI #869: Vitis-AI-RNN with Python Version 3.7 or higher?

AdrFebles, 2022-06-20:
Hello!
I'm using the Vitis-AI-RNN v1.4.1 which has Python version 3.6 installed. 
However, I also need Flower to generate the weights and this tool only supports python versions 3.7 and higher. Is there any way to use Vitis-AI-RNN with any of these Python versions?
I'll appreciate any idea!

hdtjpu, 2022-07-13:
Hi, Have you solved this issue? In [this](https://github.com/Xilinx/Vitis-AI/issues/873)  , It looks like you've solved it。

AdrFebles, 2022-07-13:
Hi @hdtjpu. Yes, we saved the weights from flower so that they would be compatible with Python 3.6 with this line:
`torch.save(model.state_dict(),` f"saved_weights/pretrained_server.pth", _use_new_zipfile_serialization=False)`

closed 2022-07-14 by AdrFebles