Xilinx/Vitis-AI #870: GPU and CPU Docker container wouldn't open on windows

Werterror, 2022-06-20:
Hi all, 

I am following one of Vitis AI Tutorials, but when I run either ./docker_run.sh xilinx/vitis-ai-gpu:latest or cpu:latest it returns:
`find: '/dev/dri' : No such file or directory
the input device is not a TTY. If you are using mintty, try prefixing the command with 'winpty'`
I also tried using mintty with "winpty" prefix and error was slightly different:
`File not found -xclmgmt*
File not found -renderD*
the input device is not a TTY. if you are using mintty, try prefixing the command with winpty`
[image]

JurgenVandendriessche, 2022-06-20:
According to the System requirements of Vitis AI, it only works on Linux ([UG1414 - System Requirements](https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Minimum-System-Requirements)).

Werterror, 2022-06-21:
@JurgenVandendriessche Indeed I missed that, Thank you.

closed 2022-06-21 by Werterror