Xilinx/Vitis-AI #871: Handling multi in/out model in PyTorch

three-cups, 2022-06-21:
Hi

I'm trying to deploy PyTorch model (.pth) into DPU.
and control DPU via VART APIs (vart::Runner).

to access specific tensor, programming manual says
'Query the DpuRunner for the shape and name of the input and output tensors'.

current model, written by PyTorch, have 4 input tensors with the same shape and slightly difference names (auto generated).
```
input_tensors[0].name = "YOLOX__YOLOX_YOLOPAFPN_backbone__CSPDarknet_backbone__Focus_stem__15889_fix"
input_tensors[0].shape = {1, 160, 320, 3}
input_tensors[1].name = "YOLOX__YOLOX_YOLOPAFPN_backbone__CSPDarknet_backbone__Focus_stem__15879_fix"
input_tensors[1].shape = {1, 160, 320, 3}
input_tensors[2].name = "YOLOX__YOLOX_YOLOPAFPN_backbone__CSPDarknet_backbone__Focus_stem__15869_fix"
input_tensors[2].shape = {1, 160, 320, 3}
input_tensors[3].name = "YOLOX__YOLOX_YOLOPAFPN_backbone__CSPDarknet_backbone__Focus_stem__15859_fix"
input_tensors[3].shape = {1, 160, 320, 3}
```

is there any way to determine tensor correspondence between python script and DpuRunner?

for example, in TensorFlow 1.x, there is useful way to naming tensors.
- tf.name_scope('xxxx')
- tf.identity(name='xxxx')
- or naming each layer ... ex. tf.nn.conv2d(..., name='xxxx')

but torch.nn.Module class don't have name attribute, and torch.nn.Identity class has no effect.

I appreciate any help, thanks

niuxjxlnx, 2022-07-27:
@three-cups :

Name in python script can not be mapped to Vitis-AI tool chain. But you can refer to the names in traced graph created by torch.jit.trace(), which have some relationship with name in Vitis-AI tool chain.

three-cups, 2022-08-03:
I tried bellow steps to two models (original float model, and quantized model generated by torch_quantizer),
but no enough information available.
1. call 'torch.jit.trace()` to model
2. save result in .pt file
3. show .pt file in Netron

This is original float model
[image]

And this is quantized model
[image]

Displayed tensor name is `2, 4, 6, 8` or `16, 32, 48, 64`, not found `15889, 15879, ...`

Is it necessary to apply this 0abd8c8f72 commit?
I use Vitis-AI 2.0, so can't apply this commit in my environment.

It seems to hard to find tensor correspondence, I'll try to remodel network.
(take slice and concat process out from network forward() method)

Thank you for your cooperation.

closed 2022-08-03 by three-cups