Xilinx/Vitis-AI #872: Vitis-ai cpu docker error

jzymessi, 2022-06-21:
I followed the steps in Readme to pull the Dockers and encountered the following problems :
1. I  use cmd  `docker pull xilinx/vitis-ai-cpu:latest`  pull docker, when i enter docker found this version is 1.4.1.978
2. I use cmd `././docker_build_cpu.sh` met the error 
[image]

hanxue, 2022-06-22:
Hi @jzymessi 

Looking at the vitis-ai-cpu Docker tags at https://hub.docker.com/r/xilinx/vitis-ai-cpu/tags,  `docker pull xilinx/vitis-ai-cpu:latest` will get you the latest 2.5.0.1260 Vitis AI v2.5 Docker. You might want to try `docker rmi xilinx/vitis-ai-cpu:latest` to delete the existing Docker image and pull again, 

Indeed `docker_build_cpu.sh` uses the outdated file https://github.com/Xilinx/Vitis-AI/blob/master/docker/dockerfiles/VERSION.txt

They will be updated later today.

hanxue, 2022-06-22:
The docker version file has been updated to 2.5.0.1260. Please can you give it a try after `git pull` your local local Vitis AI repo?

qianglin-xlnx, 2022-07-13:
Hi @jzymessi 
Is this issue solved?

jzymessi, 2022-07-26:
@qianglin-xlnx 
   Tanks,this issue is solved.

closed 2022-07-26 by jzymessi