Xilinx/Vitis-AI #873: Compilation error: Check failed: size % matmul_row_parallel == 0 size: 32, matmul_row_parallel: 64

AdrFebles, 2022-06-21:
Hello!
I launched the Vitis-AI-RNN v1.4.1 docker image by running "docker_run_rnn.sh" to quantize and compile the following LSTM autoencoder:

class Encoder(nn.Module):
    def __init__(self, seq_len, n_features, embedding_dim,device):
        super(Encoder, self).__init__()
        self.batch_size = 1
        self.device=device
        self.seq_len, self.n_features = seq_len, n_features
        self.embedding_dim, self.hidden_dim = embedding_dim, 2 * embedding_dim
        self.rnn1 = nn.LSTM(
            input_size=self.n_features,
            hidden_size=self.embedding_dim,
            num_layers=1,
            batch_first=True
        )
    def forward(self, x):
        x = x.reshape((self.batch_size, self.seq_len, self.n_features))
        x, (hn, _) = self.rnn1(x)
        return hn.reshape((self.n_features, self.embedding_dim))
     
class Decoder(nn.Module):
    def __init__(self, seq_len, input_dim, n_features):
        super(Decoder, self).__init__()
        self.batch_size = 1
        self.seq_len, self.input_dim = seq_len, input_dim
        self.hidden_dim, self.n_features = 2 * input_dim, n_features
        
        self.rnn1 = nn.LSTM(
            input_size=input_dim,
            hidden_size=self.hidden_dim,
            num_layers=1,
            batch_first=True
        )
        self.fc= nn.Linear(self.hidden_dim, n_features)

    def forward(self, x):
        x = x.repeat(self.seq_len, self.n_features)
        x = x.reshape((self.batch_size, self.seq_len, self.input_dim))
        x, _ = self.rnn1(x)
        x = x.reshape((self.seq_len, self.hidden_dim))
        out = self.fc(x)
        return out

class LSTMAutoencoder(nn.Module):
    def __init__(self, seq_len, n_features, device,embedding_dim=16):
        super(LSTMAutoencoder, self).__init__()
        self.device=device
        self.encoder = Encoder(seq_len, n_features, embedding_dim, device).to(self.device)
        self.decoder = Decoder(seq_len, embedding_dim, n_features).to(self.device)

    def forward(self, x):
        x = self.encoder(x)
        x = self.decoder(x)
        return x

I managed to quantize the model and obtained 2 Xmodel but when I execute the following command to compile the xmodels I got an error with the matmul operation.
 
(vitis-ai-rnn) Vitis-AI /workspace/>  vai_c_rnn -x quantize_result/xmodel -t DPURAHR16L -b 3 -o output

**F0621 08:23:40.218751   167 u50_inst_gen.cc:731] Check failed: size % matmul_row_parallel == 0 size: 32, matmul_row_parallel: 64**
*** Check failure stack trace: ***
    @     0x7f09e03cb4dd  google::LogMessage::Fail()
    @     0x7f09e03d3071  google::LogMessage::SendToLog()
    @     0x7f09e03caecd  google::LogMessage::Flush()
    @     0x7f09e03cc76a  google::LogMessageFatal::~LogMessageFatal()
    @     0x56143b742324  dctc::compiler::backend::u50::U50InstrGen::emitHWInstForAddWithMatMulAsInput()
    @     0x56143b742d21  dctc::compiler::backend::u50::U50InstrGen::emitHWInstForAddOrSub()
    @     0x56143b74417f  dctc::compiler::backend::u50::U50InstrGen::emitHWInstrForAddInst()
    @     0x56143b76c55e  dctc::compiler::backend::InstrGenBase::visitAddInst()
    @     0x56143b76d4af  dctc::compiler::backend::InstrGenBase::emitInstr()
    @     0x56143b70b96d  dctc::compiler::backend::u50::U50Backend::doInstrGen()
    @     0x56143b71dd91  dctc::compiler::backend::u50::U50Backend::compile()
    @     0x56143b6f272a  dctc::compiler::compile::CompilerImpl::compile()
    @     0x56143b6f3033  dctc::compiler::compile::Compiler::compile()
    @     0x56143b6c12f1  main
    @     0x7f09df366c87  __libc_start_main
    @     0x56143b6ca3f9  (unknown)
Aborted (core dumped)

When I compile the same model with DPURADR16L, the process finishes succesfully: 

(vitis-ai-rnn) Vitis-AI /workspace > vai_c_rnn -x quantize_result/xmodel -t DPURADR16L -b 1 -o output
**************************************************
* VITIS_AI RNN Compilation - Xilinx Inc.
**************************************************
INFO: Running frontend: parsing xmodel(s) into JSON representation...
INFO: JSON files will be written to output/tmp_json
[DCTC Frontend][INFO] Begin parsing Layer DecoderRnn1_StandardLstmCell_layer_0_forward.
[DCTC Frontend][INFO] Begin parsing Layer EncoderRnn1_StandardLstmCell_layer_0_forward.
[DCTC Frontend][INFO] Parsing finished, JSON files are generated in output/tmp_json
INFO: Running dctc_driver: compiling the JSON model to HW IP instructions...
INFO: dctc_driver command: 'dctc_driver -i output/tmp_json -m json -t U25 -b 1 -o output'
I0621 08:24:03.832082   177 compiler.cc:82] [INFO] Loading Json model files from path: output/tmp_json
I0621 08:24:03.833421   177 compiler.cc:84] [INFO] Parsing IR module from Json model files...
W0621 08:24:03.834378   177 json_irgen.cc:637] [WARNING] [IRGen] node will be ignored because it has not parents or children, node: actv_sgmd.
W0621 08:24:03.834466   177 json_irgen.cc:637] [WARNING] [IRGen] node will be ignored because it has not parents or children, node: actv_tanh.
W0621 08:24:03.835237   177 json_irgen.cc:637] [WARNING] [IRGen] node will be ignored because it has not parents or children, node: actv_sgmd.
W0621 08:24:03.835258   177 json_irgen.cc:637] [WARNING] [IRGen] node will be ignored because it has not parents or children, node: actv_tanh.
I0621 08:24:03.835626   177 compiler.cc:65] [INFO] Compiling model for target: U25, with batch size: 1
I0621 08:24:03.835637   177 compiler.cc:67] [INFO] Preprocessing IR module...
I0621 08:24:03.836982   177 compiler.cc:69] [INFO] Optimizing IR module...
I0621 08:24:03.839864   177 compiler.cc:71] [INFO] Compiling IR module...
I0621 08:24:03.863512   177 compiler.cc:73] [INFO] Compilation successful! The output xmodel file is located in path: output.

Could you please help me to find why I'm facing this error with the matmult operaction?
Thanks in advance

AdrFebles, 2022-06-23:
I updated the Vitis-AI repository to v2.5 and got the same error.

Mookel, 2022-06-27:
Hi, @AdrFebles thanks for your feedback.

We checked the code and found that there is indeed a bug when generating HW instructions for `Add` operator.

I am not going to cover many details here. Brief speaking, the shape of each output tensor must be aligned to meet DPU requirements, but for `DPURAHR16L` code generation part for `Add` operator, there is a mistake to get the aligned shape, thus resulting in assert fail error.

We will update it in the next release version and thanks again for your feedbacks.

AdrFebles, 2022-06-27:
Thanks for your reply @Mookel. I'm looking forward your update to compile the model on DPURAH16L.
By the way, the [lstm_pytorch example ](https://github.com/Xilinx/Vitis-AI/tree/master/src/RNN/rnn_quantizer/example/lstm_quant_pytorch) compiles succesfully with both, DPURAH16L and DPURAD16L. Why does the matmul operator work well in this case?

Mookel, 2022-06-28:
> Thanks for your reply @Mookel. I'm looking forward to your update to compile the model on DPURAH16L.
> By the way, the [lstm_pytorch example ](https://github.com/Xilinx/Vitis-AI/tree/master/src/RNN/rnn_quantizer/example/lstm_quant_pytorch) compiles succesfully with both, DPURAH16L and DPURAD16L. Why does the matmul operator work well in this case?

Hi, @AdrFebles , I guess the reason why the compiler works well for [lstm_pytorch example](https://github.com/Xilinx/Vitis-AI/tree/master/src/RNN/rnn_quantizer/example/lstm_quant_pytorch) is because its tensor size is different with your model and didn't trigger the check assertion. If you could share your `quantize_result/xmodel`, I can check it further. Thanks very much.

AdrFebles, 2022-06-28:
Thanks @Mookel, here's my [quantize_result/xmodel](https://github.com/AdrFebles/Vitis-AI-model/tree/master/quantize_result/xmodel)

Mookel, 2022-07-01:
> Thank you again @Mookel, here is my [quantize_result/xmodel](https://github.com/AdrFebles/Vitis-AI-model/tree/master/quantize_result/xmodel)

Thanks @AdrFebles  so much, I will check it and will give feedbacks ASAP.

AdrFebles, 2022-07-05:
Hi @Mookel, I am considering using Tensorflow as an alternative to Pytorch but I was wondering, if I switch to this framework, I would have the same problem with the bug in matmul operation?
I very appreciate your help.

Mookel, 2022-07-06:
> Hi, @AdrFebles , I guess the reason why the compiler works well for [lstm_pytorch example](https://github.com/Xilinx/Vitis-AI/tree/master/src/RNN/rnn_quantizer/example/lstm_quant_pytorch) is because [...]

Hi, @AdrFebles , Could you consider compiling your model using target `DPURADR16L` rather than `DPURAHR16L`?
Because our team is busying with other higher priority tasks and we need to update the conda packages after fixing the bugs. I guess there will be a delay before you can use the upgraded compiler version.

Mookel, 2022-07-12:
> Hi @Mookel unfortunately I can't use DPURADR16L due to some project requirements. That's why I was considering to switch to Tensorflow. It is possible that I'll receive the same error in this case?

Hi, @AdrFebles , sorry for the late reply. Yes, you may have same error if using TensorFlow. Hello @amd-zhidan , Could you help us to build a new conda package for RNN Compiler, thus we can build a new RNN docker image.

AdrFebles, 2022-07-14:
> Hi, @AdrFebles , sorry for the late reply. Yes, you may have same error if using TensorFlow. Hello @amd-zhidan , Could you help us to build a new conda package for RNN Compiler, thus we can build a [...]

Ok @Mookel I understand, I look forward to your news. Thank you very much.

janifer112x, 2022-07-27:
Hi, 
we have new vai_c_rnn packages  for this issue, Could you please try build new rnn docker? or if you have rnn docker already, you can try with the following steps:
inside the docker:sudo conda uninstall --force vai_c_rnn -n vitis-ai-rnn -y
cd /tmp
wget -O vai_c_rnn-1.4.1-py36h32e1ea0_5.tar.bz2 https://www.xilinx.com/bin/public/openDownload?filename=vai_c_rnn-1.4.1-py36h32e1ea0_5.tar.bz2
sudo conda install -n vitis-ai-rnn vai_c_rnn-1.4.1-py36h32e1ea0_5.tar.bz2

AdrFebles, 2022-07-27:
> Hi @janifer112x, I followed the steps and my model complied successfully. However, I got some warnings.
Thank you very much for your quick solution.
  
[image]

Mookel, 2022-07-29:
> > Hi @janifer112x, I followed the steps and my model complied successfully. However, I got some warnings.
> > Thank you very much for your quick solution.
> 
> [image]

Hi, @AdrFebles , you can just safely ignore these warnings. The nodes mentioned are redundant and have not negative effects on the code gen part.

AdrFebles, 2022-07-29:
> Hi, @AdrFebles , you can just safely ignore these warnings. The nodes mentioned are redundant and have not negative effects on the code gen part.

Ok @Mookel, thanks for replying

Mookel, 2022-08-04:
Hi, @AdrFebles, I will close issue now and please contact us if you have any other questions.

Also thanks @janifer112x so much for your help!

Hi, @qianglin-xlnx, Could you please help to close this issue, thanks so much:-).

closed 2022-08-11 by qianglin-xlnx