Xilinx/Vitis-AI #880: Vitis-AI support for other video sources

Werterror, 2022-06-22:
Hi all,
I looked through the datasheets but I couldn't find any mentions about video sources other then files stored on SD card and USB cameras. What about HDMI/Ethernet ports on ZCU102 and Kria KV260?
And J7 & J8 - IAS connectors, J9 - Raspberry Pi Camera Connector just on Kria KV260, are cameras connected to those ports are also plug and play like USB cameras, are they supported in any way?
BR

tianfang-fafafa, 2022-06-23:
hi @Werterror ,  please look into this https://docs.xilinx.com/v/u/en-US/ug1182-zcu102-eval-bd;  It may help you.

Werterror, 2022-06-27:
hi @tianfang-fafafa, since Vitis-AI is launched with PetaLinux I assumed that after instaling drivers for a device (other then USB webcam) I simply could select device when running an example with different command.

tianfang-fafafa, 2022-06-27:
hi @Werterror ， do you have any other questions? you can provide steps and err log, please

Werterror, 2022-06-27:
hi @tianfang-fafafa, I would like to know if my approach is correct since it's not described, or at least I couldn't find it, and I am not very experienced with any distro of linux. I will soon try to connect thermal camera via ethernet and usb and then I can provide logs.

Werterror, 2022-06-27:
Hi @tianfang-fafafa, with image provided for ZCU102 you can use USB camera, moreover vitis-ai examples can be run with camera input. I already have tested most of them. Now I would like to see if I can use thermal camera instead. Right now I don't have thermal camera that could be connected to the board via USB but I got couple that could be connected by the Ethernet. And I was wondering how syntax for thermal camera connected via the Ethernet or HDMI to the board would look like, or if its even possible.

tianfang-fafafa, 2022-06-27:
@Werterror ， you are right. USB camera can be ### used as input for vitis-ai example. 
[image]
but have no test for thermal camera on zcu102 board.

closed 2022-07-18 by qianglin-xlnx