Xilinx/Vitis-AI #881: DPU-TRD Vitis Flow "Found that the options configuration file contains an invalid line '['."

JH989876525, 2022-06-22:
HI guys,

I am currently working on DPU-TRD Vitis Flow with Vitis AI 2.5 on zcu102 by adding the DPU 4.0 from link below into Preferences of vitis.
https://github.com/Xilinx/Vitis-AI/tree/v2.5/reference_design
After building the application project, it shows the error message:
`ERROR: [v++ 60-1520] ***Exception: Found that the options configuration file contains an invalid line '['.`
It seems something wrong when vitis parse and paste some message into `dpu-link.cfg`. 
[image]

I finally solved this problem by removing all the comment out in `config_gui.cfg` which download from [reference_design](https://github.com/Xilinx/Vitis-AI/tree/v2.5/reference_design).

Hope this message could help some others.

jiaz-xlnx, 2022-06-23:
Thanks @JH989876525 I fix this issue now. the Vtis will add some issues in the dpu-link.cfg file. when analysis the config_gui.cfg file. it is a workaround to delete all the comment.

jiaz-xlnx, 2022-08-26:
hi @JH989876525 We have fixed this issue and updated the design.

closed 2022-08-31 by qianglin-xlnx