Xilinx/Vitis-AI #882: VAI TRACE: Confusing profiling results

AfifaIshtiaq, 2022-06-22:
First I run vaitrace profiler in normal mode and I get the follwing results
XAIEFAL: INFO: Resource group Generic is created.
Starting 2 threads...
else batch_size 540
else indices 540 0
lenth of list_ids_temp 540
INFO:root:vaitrace timeout, waiting for Python thread terminated
rx_data_input Shape: (5400, 14, 72, 2)
if batch_size 540
rx_data_input Shape: (5400, 14, 72, 2)
dpu_inputs [<xir.Tensor named 'quant_RX-Data-In'>, <xir.Tensor named 'quant_TX-Pilot-In'>, <xir.Tensor named 'quant_Raw-Channel-Est-In'>]
DPU runner input: quant_RX-Data-In  Shape: [5, 14, 72, 2]
DPU runner input: quant_TX-Pilot-In  Shape: [5, 14, 72, 2]
DPU runner input: quant_Raw-Channel-Est-In  Shape: [5, 14, 72, 2]
DPU runner output: quant_Conv-Out_relu_fix  Shape: [5, 14, 72, 2]
Done init_dpu_runner
ThreadbatchSize 5
n_of_Radio_Frames 5400
write_index 0
dpu_inputs [<xir.Tensor named 'quant_RX-Data-In'>, <xir.Tensor named 'quant_TX-Pilot-In'>, <xir.Tensor named 'quant_Raw-Channel-Est-In'>]
DPU runner input: quant_RX-Data-In  Shape: [5, 14, 72, 2]
DPU runner input: quant_TX-Pilot-In  Shape: [5, 14, 72, 2]
DPU runner input: quant_Raw-Channel-Est-In  Shape: [5, 14, 72, 2]
DPU runner output: quant_Conv-Out_relu_fix  Shape: [5, 14, 72, 2]
Done init_dpu_runner
ThreadbatchSize 5
n_of_Radio_Frames 5400
write_index 5400
------------------------------------
**Throughput=1009.56 fps, total frames = 10800, time=10.6977 seconds**
INFO:root:Generating VTF
INFO:root:Processing xmodel information
INFO:root:[Summary] DPU cores involved: 1
INFO:root:[Summary] Total inferences: 10800
INFO:root:[Summary] Overall inferences/second: 1010.17
INFO:root:[vart_trace.csv]: Events number: 4320

But when I run it with finegrained mode its throughput is halved why is that?
XAIEFAL: INFO: Resource group Generic is created.
Starting 2 threads...
else batch_size 540
else indices 540 0
lenth of list_ids_temp 540
INFO:root:vaitrace timeout, waiting for Python thread terminated
rx_data_input Shape: (5400, 14, 72, 2)
if batch_size 540
rx_data_input Shape: (5400, 14, 72, 2)
dpu_inputs [<xir.Tensor named 'quant_RX-Data-In'>, <xir.Tensor named 'quant_TX-Pilot-In'>, <xir.Tensor named 'quant_Raw-Channel-Est-In'>]
DPU runner input: quant_RX-Data-In  Shape: [5, 14, 72, 2]
DPU runner input: quant_TX-Pilot-In  Shape: [5, 14, 72, 2]
DPU runner input: quant_Raw-Channel-Est-In  Shape: [5, 14, 72, 2]
DPU runner output: quant_Conv-Out_relu_fix  Shape: [5, 14, 72, 2]
Done init_dpu_runner
ThreadbatchSize 5
n_of_Radio_Frames 5400
write_index 0
dpu_inputs [<xir.Tensor named 'quant_RX-Data-In'>, <xir.Tensor named 'quant_TX-Pilot-In'>, <xir.Tensor named 'quant_Raw-Channel-Est-In'>]
DPU runner input: quant_RX-Data-In  Shape: [5, 14, 72, 2]
DPU runner input: quant_TX-Pilot-In  Shape: [5, 14, 72, 2]
DPU runner input: quant_Raw-Channel-Est-In  Shape: [5, 14, 72, 2]
DPU runner output: quant_Conv-Out_relu_fix  Shape: [5, 14, 72, 2]
Done init_dpu_runner
ThreadbatchSize 5
n_of_Radio_Frames 5400
write_index 5400
------------------------------------
Throughput=457.78 fps, total frames = 10800, time=23.5921 seconds
INFO:root:Generating VTF
INFO:root:Processing xmodel information
INFO:root:[Summary] DPU cores involved: 1
INFO:root:[Summary] Total inferences: 57600
INFO:root:[Summary] Overall inferences/second: 27401.87
INFO:root:[vart_trace.csv]: Events number: 23040

li900309, 2022-06-24:
when the fine grained profiling model is enabled, models will be run subgraph-by-subgraph.
1. It will introduce significant scheduling and memory access overhead, especially for small models.
2. Some xcompiler's optimizations will be turned-off.

qianglin-xlnx, 2022-07-18:
Hi @AfifaIshtiaq 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-07-18 by qianglin-xlnx