Xilinx/Vitis-AI #883: Missing libprotobuf after updating Vitis AI Runtime on ZCU102

cl600class, 2022-06-23:
I have followed the instructions in Step 2.4 [here](https://github.com/Xilinx/Vitis-AI/tree/master/setup/mpsoc), but I got the error message after importing vart.
[image]

Should I re-install the board image in this case?

qianglin-xlnx, 2022-06-23:
Hi @cl600class 
Which Petalinux version do you use? 2022.1 or 2021.2?
And the vart version you install before is?

cl600class, 2022-06-23:
Hi, 
Currently I'm using v2021.1 since Vitis AI 1.4.
The original vart version is 2021.2, and the issue occurs when upgrading to 2022.1.
Thanks.

xihuazhanghui, 2022-06-23:
Hi, @cl600class 

You should upgrade your board image to 2022.1 too. the Petalinux sdk version must match with board image version

Thanks

qianglin-xlnx, 2022-07-18:
Hi @cl600class 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-07-18 by qianglin-xlnx

cl600class, 2022-07-18:
Hi @qianglin-xlnx ,
Sorry for the late repy. The problem has been solved. Thanks for your help.