Xilinx/Vitis-AI #884: export xmodel failed when use vai_q_pytorch

JackzhangLOVEcode, 2022-06-23:
hey, guys, i have completed calib mode, but when run test mode, TypeError come up, here are information: 
[VAIQ_NOTE]: =>Converting to xmodel ...
Traceback (most recent call last):
  File "quantize_zc.py", line 147, in <module>
    run_main()
  File "quantize_zc.py", line 140, in run_main
    quantize(args.build_dir,args.quant_mode,args.batchsize)
  File "quantize_zc.py", line 95, in quantize
    quantizer_en.export_xmodel(deploy_check=False, output_dir=quant_model_en)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/apis.py", line 127, in export_xmodel
    self.processor.export_xmodel(output_dir, deploy_check)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/qproc/base.py", line 248, in export_xmodel
    dump_xmodel(output_dir, deploy_check, self._lstm_app)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/qproc/base.py", line 321, in dump_xmodel
    deploy_graphs = get_deploy_graph_list(quantizer.quant_model, quantizer.Nndctgraph)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/qproc/utils.py", line 276, in get_deploy_graph_list
    g_optmizer = DevGraphOptimizer(nndct_graph)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/nndct_shared/compile/deploy_optimizer.py", line 40, in __init__
    self._dev_graph.clone_from(nndct_graph)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/nndct_shared/nndct_graph/base_graph.py", line 105, in clone_from
    self._top_block.clone_from(src_graph.block, local_map, converted_nodes)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/nndct_shared/nndct_graph/base_block.py", line 47, in clone_from
    self.append_node(self.owning_graph.create_node_from(node, local_map, converted_nodes))
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/nndct_shared/nndct_graph/base_graph.py", line 132, in create_node_from
    node.clone_from(src_node, local_map)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/nndct_shared/nndct_graph/base_node.py", line 119, in clone_from
    self.op.clone_from(src_node.op, local_map)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/nndct_shared/nndct_graph/base_operator.py", line 221, in clone_from
    setattr(self, config, new_value)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/parse/torch_op_def.py", line 203, in padding
    self.set_attr(self.AttrName.PAD, [value[0], value[0]])
TypeError: 'int' object is not subscriptable

i have no idea about the error, anyone can help? thank you for any advise

wangxd-xlnx, 2022-07-07:
Hi @JackzhangLOVEcode 

Could you provide some more details such as which model did you use and if you use Vitis-AI docker or your environment info?

JackzhangLOVEcode, 2022-07-08:
> Hi @JackzhangLOVEcode
> 
> Could you provide some more details such as which model did you use and if you use Vitis-AI docker or your environment info?

thanks for reply, I have solved this problem, the error is caused by the one-dimensional convolution function, I have changed it to two-dimensional convolution function.

closed 2022-07-11 by niuxjxlnx

zyt1024, 2023-08-17:
> thanks for reply, I have solved this problem, the error is caused by the one-dimensional convolution function, I have changed it to two-dimensional convolution function.

Hello, I also encountered this problem in my network where I used Conv1D, but I'm using Vitis-AI 2.5. I would like to know if Conv1D is not supported in Vitis-AI ？

zyt1024, 2024-01-08:
> Hello, I also encountered this problem in my network where I used Conv1D, but I'm using Vitis-AI 2.5. I would like to know if Conv1D is not supported in Vitis-AI ？

@JackzhangLOVEcode   Hello, I also encountered this problem in my network where I used Conv1D, but I'm using Vitis-AI 2.5. I would like to know if Conv1D is not supported in Vitis-AI ？