Xilinx/Vitis-AI #885: infer_shape AssertionError when compiling a custom network with Vitis-AI 1.4 to 2.5

TitechTraj, 2022-06-23:
I have a custom network that combines 3 CNNs. 2 are parallel then their outputs get concatenated, reshaped, and fed to a 3rd CNN.
I used to work with Vitis AI 1.3 and it runs without issues. Since the update to 1.4, I keep having this error during compilation (even when trying with the new version 2.5)  
```
[INFO] tensorflow2 model: /workspace/TF2_DeepSense/quantized_model.h5
[INFO] keras version: 2.8.0
[INFO] Tensorflow Keras model type: functional
[INFO] parse raw model     :100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 27/27 [00:00<00:00, 13086.00it/s]               
[INFO] infer shape (NHWC)  : 59%|████████████████████████████████████████████████████████████████████████████                                                    | 22/37 [00:00<00:00, 26206.95it/s]               
Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/bin/xnnc-run", line 33, in <module>
    sys.exit(load_entry_point('xnnc==2.5.0', 'console_scripts', 'xnnc-run')())
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/__main__.py", line 49, in main
    runner.normal_run(args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/runner.py", line 123, in normal_run
    target=target,
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/xconverter.py", line 145, in run
    model_files, model_type, _layout, in_shapes, batchsize
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/core.py", line 123, in make_xmodel
    model_type=model_t,
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py", line 107, in to_xmodel
    model_type,
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py", line 177, in create_xmodel
    name, layers, layout, in_shapes, batchsize
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py", line 2366, in __create_xmodel_from_tf2
    if not xmodel.infer_shape(Layout.NHWC):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/ir/xmodel.py", line 506, in infer_shape
    ok, error = self.__do_shape_inference(Layout[self.layout])
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/ir/xmodel.py", line 531, in __do_shape_inference
    xnode.infer_shape(layout)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/ir/xnode.py", line 3629, in infer_shape
    assert 0 < len(self.shape) <= 4
AssertionError

```
The quantization works fine, I get the error during compilation.
I assume that the compiler always expects a shape of 4 dimensions. The issue is probably caused by the reshape and concatenation that add a new dimension :

CNN1 and CNN2 have a reshape: 1,20,4,64 to 1,20,4,1,64
Then a concat layer merges them to: 1,20,4,2,64 and reshapes them back to 1,20,8,64 to be used by CNN3.

This didn't raise any error with Vitis 1.3
What changed in the compiler since 1.4? and how to bypass this issue?

zhenzhen-AMD, 2022-06-24:
> 

Hi @TitechTraj ,
Please provide the floating point model and the quantized model and the code for the quantized model. thanks.

TitechTraj, 2022-06-24:
Hi @zhenzhen-AMD 
Thank you for the quick reply.

Here are my floating point model and the quantized model:

[myCNN.zip](https://github.com/Xilinx/Vitis-AI/files/8972867/myCNN.zip)
[quantized_model.zip](https://github.com/Xilinx/Vitis-AI/files/8972868/quantized_model.zip)

I used the following code for quantization

```
calib_dataset = np.load("calibration_data/inputs.npy")
float_model = tf.keras.models.load_model("myCNN.h5")
from tensorflow_model_optimization.quantization.keras import vitis_quantize
quantizer = vitis_quantize.VitisQuantizer(float_model)
quantized_model = quantizer.quantize_model(calib_dataset=calib_dataset)
quantized_model.save('quantized_model.h5')
```

TitechTraj, 2022-07-22:
Thank you very much!

zhenzhen-AMD, 2022-07-29:
Hi @TitechTraj ,

The patch has been generated. Please use the following command to update:
wget -O xnnc-2.5.0-py36hd51751d_201.tar.bz2  https://www.xilinx.com/bin/public/openDownload?filename=xnnc-2.5.0-py36hd51751d_201.tar.bz2
wget -O xnnc-2.5.0-py37hd51751d_201.tar.bz2  https://www.xilinx.com/bin/public/openDownload?filename=xnnc-2.5.0-py37hd51751d_201.tar.bz2
inside docker. : 
conda update -n vitis-ai-tensorflow xnnc-2.5.0-py36hd51751d_201.tar.bz2
conda update -n vitis-ai-tensorflow2 xnnc-2.5.0-py37hd51751d_201.tar.bz2

TitechTraj, 2022-07-31:
Thank you for the patches, I installed them but I get this error when compiling:
`[UNILOG][FATAL][XCOM_INVALID_TARGET][Invalid DPU target] Compiler backend cannot support DPUCZDX8G_ISA0. (valid list: DPUCADF8H_ISA0 DPUCAHX8H_ISA2 DPUCAHX8L_ISA0 DPUCVDX8G_ISA3 DPUCVDX8H_ISA1 DPUCZDI4G_ISA0 DPUCZDX8G_ISA1)`

I suppose the fingerprint format of the DPU has changed? I need to recreate a new dpu design using Vitis/Vivado 2022.1?

wanghong4compiler, 2022-08-03:
Hi @TitechTraj ,
According to the error message, the target name should be DPUCZDX8G_ISA1 instead of DPUCZDX8G_ISA0.
Yes, it you switch to 2.5 from 1.4, the fingerprint and ISA are both upgraded, so please check if your dpu's fingerprint is match, if not, please recreate a new one.

TitechTraj, 2022-08-06:
It seems that it works. I was able to compile the model without errors using the new fingerprint generated with the new DPU IP 4.0 provided with Vitis AI 2.5
Concerning the compilation, I think this issue is solved. I will check if the model runs on the board as expected.
Thank you very much for your support!

closed 2022-08-06 by TitechTraj