Xilinx/Vitis-AI #887: Tensorflow2 Compile error with input concatenation layer

Daniel-O551, 2022-06-24:
Hi unfortunately I get the following error when compiling my model.
`[UNILOG][FATAL][XCOM_ITEM_UNDEFINED][The requested item was not found or defined!]`

Judging from my testing, the compiler is having problems with my concatenated layer, however I can't find an error in my code.
I have attached a corresponding test model including quantization, in which the error occurs.

My compile command:
```Shell
vai_c_tensorflow2 -m quantized.h5 -a /opt/vitis_ai/compiler/arch/DPUCVDX8G/VCK190/arch.json -o output -n netname
```
Python code:
```Python
#!/usr/bin/env python
# coding: utf-8

import pathlib
import pandas as pd

import tensorflow as tf
from sklearn.model_selection import train_test_split
from sklearn.datasets import make_regression
from sklearn.preprocessing import StandardScaler

# In[38]:

X, y = make_regression(n_samples=1000, n_features=20, noise=0.5, random_state=42)
X = StandardScaler().fit_transform(X)
y = StandardScaler().fit_transform(y.reshape(len(y),1))[:,0]
X_train, X_test, y_train, y_test = train_test_split(X, y, test_size=0.25, random_state=42)

inp_list=[tf.keras.layers.Input(shape=(1,), name=str(i)) for i in range(20)]  
cat_layer = tf.keras.layers.Concatenate()(inp_list)
dense_0 = tf.keras.layers.Dense(10, activation='relu')(cat_layer)
dense_1 = tf.keras.layers.Dense(1, activation='linear')(dense_0)
model = tf.keras.models.Model(inp_list, dense_1)

model.compile(
    optimizer=tf.keras.optimizers.Adam(learning_rate=.01),
    loss='mean_squared_error',
)

X_train = pd.DataFrame(X_train)
X_train.columns =[str(i) for i in X_train.columns] 
X_test = pd.DataFrame(X_test)
X_test.columns =[str(i) for i in X_test.columns] 
history = model.fit(dict(X_train), y_train, epochs=50, verbose=2, validation_split=0.2)
hist = pd.DataFrame(history.history)
hist['epoch'] = history.epoch

print(model.summary())

tf.keras.models.save_model(model, 'concat_model.h5')
pd.DataFrame(X_test).to_csv('concat_model/X.csv', index=False)
pd.DataFrame(y_test).to_csv('concat_model/y.csv', index=False)

# In[45]:

model_dir = pathlib.Path("concat_model")
model = tf.keras.models.load_model("concat_model.h5")

#X = pd.read_csv(model_dir / "X.csv")
X = pd.DataFrame(X)
X.columns =[str(i) for i in X.columns] 
#y = pd.read_csv(model_dir / "y.csv")

print(f"mean_squared_error: {model.evaluate(dict(X), y)}")

from tensorflow_model_optimization.quantization.keras import vitis_quantize

model = tf.keras.models.load_model("concat_model.h5")

quantizer = vitis_quantize.VitisQuantizer(model)
calib_dataset = dict(X.iloc[:10])

quantized_model = quantizer.quantize_model(calib_dataset=calib_dataset, include_cle=True, cle_steps=10, include_fast_ft=True)
quantized_model.save("quantized.h5")
quantized_model.compile(loss=tf.keras.losses.MeanSquaredError())
quantized_model.evaluate(dict(X), y)
quantizer.dump_model(quantized_model, dataset=dict(X.iloc[0:1]), dump_float=True)
```

Daniel-O551, 2022-07-08:
Hi, can I help you with troubleshooting in any way? Feel free to contact me if you have problems reproducing the error.

sh39o, 2022-08-10:
Hi, the compiler didn't consider concatenated layer so many inputs, and the problem is fixed in the new released patch.
please refer to for details: https://github.com/Xilinx/Vitis-AI/blob/master/docs/quick-start/install/Vitis%20AI%202.5%20Aug%202022%20Patch.md.

closed 2022-08-31 by qianglin-xlnx