Xilinx/Vitis-AI #888: Error with replace_pytorch.sh with Docker 2.5 version

chuong98, 2022-06-24:
[image]
Hello, I can create new Docker Vitis 2.5 version, but then there is still error when replacing pytorch. Please fix

chuong98, 2022-06-25:
I manage to fix this error by replacing the link in the `replace_pytorch.sh`
`wget -O conda-channel.tar.gz --progress=dot:mega https://www.xilinx.com/bin/public/openDownload?filename=conda-channel_2.5.0.1260-01.tar.gz`

niuxjxlnx, 2022-07-05:
@chuong98:

Thanks for your finding. We will update the script in vitis-ai github first.

niuxjxlnx, 2022-07-11:
The script is updated at this link: 
https://github.com/Xilinx/Vitis-AI/blob/master/docker/dockerfiles/replace_pytorch.sh

closed 2022-07-11 by niuxjxlnx