Xilinx/Vitis-AI #889: how to deploy and run target from model zoo

watchara-knot, 2022-06-24:
I download the model from https://github.com/Xilinx/Vitis-AI/tree/master/model_zoo/model-list/pt_OFA-rcan_DIV2K_360_640_45.7G_2.5

and download name: pt_OFA-rcan_DIV2K_360_640_45.7G_2.5 lean and follow step get only ,xmodel and .json, my question is how to deploy and run target?

--

I knew can download xmodel 

"name: ofa_rcan_latency_pt

 type: xmodel

 board: zcu102 & zcu104 & kv260

 download link: https://www.xilinx.com/bin/public/openDownload?filename=ofa_rcan_latency_pt-zcu102_zcu104_kv260-r2.5.0.tar.gz

 checksum: 8be5e1cf6cabef8f974f2522a3b0d604"

to test in the example in Vitis (Vitis library ), but step in gpu not provide .prototext

qianglin-xlnx, 2022-06-25:
Hi @watchara-knot 
You can refer to https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Quick-Start to deploy the model from model zoo.
For the model prototxt file, you can copy it from  https://www.xilinx.com/bin/public/openDownload?filename=ofa_rcan_latency_pt-zcu102_zcu104_kv260-r2.5.0.tar.gz and modify it .
You can also refer to https://docs.xilinx.com/r/en-US/ug1354-xilinx-ai-sdk/Using-the-Configuration-File

watchara-knot, 2022-06-25:
@qianglin-xlnx Thank you very to fast response, I will try to learn it.

One more question is https://github.com/Xilinx/Vitis-AI/tree/v2.0/demo/Vitis-AI-Library/samples/rcan

Which one do I have to refer to in case of If I want to change the output size image in the target board?

please let me know.

closed 2022-06-26 by watchara-knot