Xilinx/Vitis-AI #890: [Invalid DPU target] Error when compile xmodel with Vitis 2.5 version for ZCU104,

chuong98, 2022-06-25:
Hello, I can compile a pytorch model into int.xmodel, however, in the last step, when compile to `xmodel` with `vai_c_xir`, it throws error that the device is not supported.
[image]

Here is the fingerprint of ZCU104
```

//use xdputil query  
{
    "fingerprint":"0x1000020f6014407"
}
```

JackzhangLOVEcode, 2022-06-26:
compile with VAI2.0

chuong98, 2022-06-26:
I am using Docker, so I expect the default libs should work. 

Do you mean to build another Docker version 2.0 to compile the model ?

JackzhangLOVEcode, 2022-06-27:
> I am using Docker, so I expect the default libs should work.
> 
> Do you mean to build another Docker version 2.0 to compile the model ?
I got same error when compiled models for ZCU102 by vitis2.5, but when checkout to vitis2.0 by ./docker_run.sh xilinx/vitis-ai-cpu:2.0.0.1103, it work fine, you can try it

chuong98, 2022-06-28:
Do i need to upgrade the firmw

wanghong4compiler, 2022-06-28:
Hi @chuong98 ,
yes, if you switch to Vitis 2.5, all components needs to be consistent.

wanghong4compiler, 2022-06-28:
Hi @chuong98 ,
sorry, my bad, please try "0x0101000016010407" again, sorry again.

chuong98, 2022-06-28:
Thanks, it works

closed 2022-06-28 by chuong98