Xilinx/Vitis-AI #891: [XCOME_VALUE_UNMATCH]: vai_c_xir compiled error from pytorch model

JackzhangLOVEcode, 2022-06-26:
I quantified two models, when compile them by vai_c_xir, one was ok, another compiled failed, here is error information
[image]
Anyone knows what the error information means? All suggestions are welcome!!

wanghong4compiler, 2022-06-27:
Hi @JackzhangLOVEcode ,
Could you share the model file build/quant_model_en/Encoder_int.xmodel, so that I can reproduce this issue?

JackzhangLOVEcode, 2022-06-27:
> 

OK ！ Encoder_int.xmodel compiled fine,  Decoder_int.xmodel compiled failed,  here is it，thank you for reply
[Decoder_int.zip](https://github.com/Xilinx/Vitis-AI/files/8991704/Decoder_int.zip)

sh39o, 2022-08-10:
Hi @JackzhangLOVEcode 
The bug has been fixed in the new patch https://github.com/Xilinx/Vitis-AI/blob/master/docs/quick-start/install/Vitis%20AI%202.5%20Aug%202022%20Patch.md. 
"Fixed a compiler bug to support concatenate operator cascaded with multiple reshape operators" 
Thank you for reporting the bug!

closed 2022-08-31 by qianglin-xlnx