Xilinx/Vitis-AI #896: The pruning method used in person-reid.

LorenzoSun-V, 2022-06-29:
In vitis-ai 2.5, person-reid project has been update. And I saw new models and kernels in the code like ResNeXt and IBN. Does this mean model whose backbone is ResNeXt-IBN can be prune and quantized in vitis-ai 2.5? If so, which pruning method do you use? One_step,  iterative pruning or OFA?

wenjingk-xilinx, 2022-07-05:
HI @LorenzoSun-V ,

Currently only Resnet backbone is supported.  The released resnet-based pruned models are pruned using iterative pruning method.

jia0511, 2022-07-06:
HI @wenjingk-xilinx 

I would like to consult some details of the current iterative pruning ReID. Compared with the pruning in vitis-ai2.5 modelzoo, 70% of the pruning has dropped a few points. We want to reproduce your results. You iteratively step by step during the iterative pruning process. What is the process of the ratio, such as 0.25-0.5-0.7, how many epochs are retrained after each pruning, and is the last time the most? What is the hyperparameter information for each pruning training? Is the sparse model or the extracted slim model used for secondary pruning after pruning?

wenjingk-xilinx, 2022-07-07:
Hi @jia0511 ,

I do iteratively pruning with ratios 0.4-0.5-0.6-0.7. For each pruning, I finetuned about 80 epochs, with the base_lr = 0.1*base_lr_of_baseline_model, the other training configurations stay the same as the baseline model. The last epoch doesn't necessarily get the best model. The pruned sparse model is used as the pre-trained model for the next step of pruning.

jia0511, 2022-07-07:
HI @wenjingk-xilinx,

Thanks for your reply, I would like to know if 80epoch is the total number of fine-tuning, 20 epoch at a time,  or 80 epoch fine-tuning after each pruning?

wenjingk-xilinx, 2022-07-07:
HI @jia0511 
For each pruning ratio, I finetuned about 80 epochs.

jia0511, 2022-07-07:
HI @wenjingk-xilinx,

Thank you very much.

closed 2022-07-11 by LorenzoSun-V