Xilinx/Vitis-AI #898: ZCU102 Failed to fetch URL

x0Y0x, 2022-06-30:
I tried building Petalinux 2021.2 with the vitis-ai packagegroup but got stuck with the following error messages. It seems like it is unable to fetch the necessary repositories. This error looks similar to this one  #861.

`WARNING: unilog-2.0-r0 do_fetch: Failed to fetch URL git://gitenterprise.xilinx.com/Vitis/vitis-ai-staging;protocol=https;branch=2.0, attempting MIRRORS if available
WARNING: xir-2.0-r0 do_fetch: Failed to fetch URL git://gitenterprise.xilinx.com/Vitis/vitis-ai-staging;protocol=https;branch=2.0, attempting MIRRORS if available`

tianfang-fafafa, 2022-07-04:
hi @x0Y0x ,   The SRC of recipe should be updated after Vitis-AI release, because the commit would be confirmed after Vitis-AI release. So please look into this https://github.com/Xilinx/Vitis-AI/commit/fc74d404563d9951b57245443c73bef389f3657f which is behind of tag-2.0

tianfang-fafafa, 2022-07-20:
hi @x0Y0x 
have you solved this issue?
Since we haven’t received your reply for a long time, we assume you have solved this issue and I’m going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-07-20 by qianglin-xlnx