Xilinx/Vitis-AI #899: downloader.py issue

AfifaIshtiaq, 2022-07-01:
when I run downloader.py I get the following issue
File "/home/vitis-ai-user/petalinux_sdk_2021.2/sysroots/x86_64-petalinux-linux/usr/lib/python3.8/http/client.py", line 1424, in connect
    self.sock = self._context.wrap_socket(self.sock,
  File "/home/vitis-ai-user/petalinux_sdk_2021.2/sysroots/x86_64-petalinux-linux/usr/lib/python3.8/ssl.py", line 500, in wrap_socket
    return self.sslsocket_class._create(
  File "/home/vitis-ai-user/petalinux_sdk_2021.2/sysroots/x86_64-petalinux-linux/usr/lib/python3.8/ssl.py", line 1040, in _create
    self.do_handshake()
  File "/home/vitis-ai-user/petalinux_sdk_2021.2/sysroots/x86_64-petalinux-linux/usr/lib/python3.8/ssl.py", line 1309, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1123)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "downloader.py", line 158, in <module>
    main()
  File "downloader.py", line 156, in main
    download(loadLinkList)
  File "downloader.py", line 118, in download
    urllib.request.urlretrieve(load_Link, name,Schedule)
  File "/home/vitis-ai-user/petalinux_sdk_2021.2/sysroots/x86_64-petalinux-linux/usr/lib/python3.8/urllib/request.py", line 247, in urlretrieve
    with contextlib.closing(urlopen(url, data)) as fp:
  File "/home/vitis-ai-user/petalinux_sdk_2021.2/sysroots/x86_64-petalinux-linux/usr/lib/python3.8/urllib/request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
  File "/home/vitis-ai-user/petalinux_sdk_2021.2/sysroots/x86_64-petalinux-linux/usr/lib/python3.8/urllib/request.py", line 525, in open
    response = self._open(req, data)
  File "/home/vitis-ai-user/petalinux_sdk_2021.2/sysroots/x86_64-petalinux-linux/usr/lib/python3.8/urllib/request.py", line 542, in _open
    result = self._call_chain(self.handle_open, protocol, protocol +
  File "/home/vitis-ai-user/petalinux_sdk_2021.2/sysroots/x86_64-petalinux-linux/usr/lib/python3.8/urllib/request.py", line 502, in _call_chain
    result = func(*args)
  File "/home/vitis-ai-user/petalinux_sdk_2021.2/sysroots/x86_64-petalinux-linux/usr/lib/python3.8/urllib/request.py", line 1393, in https_open
    return self.do_open(http.client.HTTPSConnection, req,
  File "/home/vitis-ai-user/petalinux_sdk_2021.2/sysroots/x86_64-petalinux-linux/usr/lib/python3.8/urllib/request.py", line 1353, in do_open
    raise URLError(err)
urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1123

wangxd-xlnx, 2022-07-05:
Hi @AfifaIshtiaq 

Could you provide your environment info, such as if you're in Vitis-AI docker or not? 
We cannot reproduce your error. I found some solutions on the Internet, maybe you can refer to it.

To solve urllib.error.URLError: urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate, you just need to Install Certificates.command. Just follow this step In macOS just open Macintosh HD Now Select Application Then Select Python folder ( Python3.6, Python3.7 Whatever You are using just select this folder ) Then, double click on Install Certificates.command. Now your error should be solved.

https://exerror.com/urllib-error-urlerror-urlopen-error-ssl-certificate_verify_failed-certificate-verify-failed-unable-to-get-local-issuer-certificate/

qianglin-xlnx, 2022-08-23:
Hi @AfifaIshtiaq 
Is this issue solved?

qianglin-xlnx, 2022-08-31:
Closing since no activity for more than 1 week, please reopen if you still have question, thanks

closed 2022-08-31 by qianglin-xlnx