Xilinx/Vitis-AI #900: how to use DPU

easilee, 2022-07-04:
I have generated the DPU file, how can I use it to do inference?ZCU-102

easilee, 2022-07-04:
someone let me use app_mt.py, they say you can copy your val.py (inference code)

easilee, 2022-07-11:
> 你好@easilee 可以参考https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Quick-Start 也可以参考： [https](https://github.com/Xilinx/Vitis-AI-Tutorials) ://github.com/Xilinx/Vitis-AI-Tutorials .

 I've generated DPU，my network is U-net，Input size 1*3*512*512，but use DPU show NAN

easilee, 2022-07-11:
> Hi @easilee You can refer to https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Quick-Start You can also refer to: https://github.com/Xilinx/Vitis-AI-Tutorials .

When DPU is used, the maximum and minimum values of InputData are NAN

qianglin-xlnx, 2022-07-13:
Same issue, refer to : https://github.com/Xilinx/Vitis-AI/issues/913