Xilinx/Vitis-AI #901: Unable to generate arch.json or .dcf from .hwh file - Vitis-AI. Docker image version: 2.5.0.1260

anirudhitagi, 2022-07-04:
Greetings,

I am using an ultra96 v2 board I am trying to generate a .dcf and subsequently an arch.json file with custom DPU configurations:  
<PARAMETER NAME="DNNDK_PRINT" VALUE="Number of DPU Cores:1;Arch of DPU:B1600;RAM Usage:High;Channel Augmentation:Enabled;DepthWiseConv:Enabled;AveragePool:Enabled;ReLU Type:ReLU + LeakyReLU + ReLU6;Number of SFM cores:1;S-AXI Clock Mode:Independent;dpu_2x Clock Gating:Enabled;DSP48 Maximal Cascade Length:4;DSP48 Usage:High;Ultra-RAM Use per DPU:0;Enable timestamp auto-update:Enabled;Target Version:1.4.1;AXI Protocol:AXI4;S-AXI Data Width:32;M-AXI GP Data Width:32;M-AXI HP Data Width (DPU):128;M-AXI HP Data Width (SFM):128;M-AXI ID Width:2;DSP Slice Count:326;Ultra-RAM Count:0.0;Block-RAM Count:167.5"/>

I used Vivado 2021.2 to create the block design with custom DPU configs (as above) and generated the system.hwh (attached as system.zip)
[system.zip](https://github.com/Xilinx/Vitis-AI/files/9036853/system.zip)

However, when I attempt to generate .dcf from dlet in the latest (As of 2022-07-04) VITIS-AI-CPU (Docker image version - 2.5.0.1260), I get the following error: **bash: "dlet": command not found**

I suspected that dlet may not be bundled in this latest Vitis-Ai, so I pulled an earlier version of Vitis AI namely: 1.2.82
Here dlet command is working, because when I run dlet -v I see this:
**DLet version 1.0
Build Label: Jun 23 2020 03:34:15
Copyright @2019 Xilinx Inc. All Rights Reserved.**

However, when I run dlet -f system.hwh command I get the following error:
**[DLet]Failed to find DPU parameters from DPU HWH.**

1. Can someone please help identify the issue here? Is there a problem my .hwh file, dlet version or vitis-ai version?
2. If this is not solve able can you please help me the a DPU Fingerprint which has the DPU configurations as mentioned above (I am mainly focused on enabling a SoftMax core)? 
3. Is there a list of DPU Configurations and their fingerprints listed somewhere? I could not find any link or documentation listing the finger prints.

Please let me know if I need to provide any more information. 

Thanks and Regards,
-AI

davidxudong, 2022-07-05:
Hi @anirudhitagi 
1. We won't use the dlet in the future，please update to the lastest Vitis-AI in https://github.com/Xilinx/Vitis-AI
2. To generate arch.json, please refer to the lastest DPUCZDX8G reference design in https://github.com/Xilinx/Vitis-AI/tree/master/reference_design, and follow the readme.
3. There is no list that you mentioned.

Thanks

anirudhitagi, 2022-07-05:
Hi @davidxudong,

Thank you for your response. I am able to generate the arch.json file with the desired DPU configurations.
I was working with older DPU V3.4, but now after updating to the latest DPU v4.0 I am able to configure to the right version of DPU.

Thanks again!

closed 2022-07-05 by anirudhitagi