Xilinx/Vitis-AI #902: Documentation Incorrect

ajith-xilinx, 2022-07-04:
After "git clone" information on main landing page, mention about host setup & card setup. 

```cd ./setup/vck5000/ && source ./install.sh```

Without this step, the mentioned process will throw error in later steps.

SuZhan0322, 2022-07-19:
This step has been pointed out in https://github.com/Xilinx/Vitis-AI/tree/master/setup/vck5000#setup-versal-accelerator-card.  If I noticed the wrong location, please help to point it out, thanks.

SuZhan0322, 2022-08-07:
Hi @ajith-xilinx , 
Is there any comment update to this issue? thanks

ajith-xilinx, 2022-08-19:
The Main Landing Page : https://github.com/Xilinx/Vitis-AI 

In this page, there is no mention about pre-requisites ( XRT Installation & DSA flash etc. ) 

The Installation process according to Main landing page, start from **git clone** & **docker pull**. 
Without installing XRT, this **docker run** will fail. 

So, it has to fist point to : 
```
cd ./setup/<device>   # cd ./setup/vck5000
sudo ./install.sh
```

And then, **git clone** -> **docker pull** -> **docker run**

SuZhan0322, 2022-09-01:
Hi @ajith-xilinx ,
Thanks for your great suggestion, we will clarify this step in the next release.

closed 2022-09-01 by SuZhan0322