Xilinx/Vitis-AI #905: About the inspector of the vai_q_pytorch

Truthonlyone, 2022-07-05:
Hello,
I tried to inspect float model before quantization, and modified the resnet18_quant.py to adapt the cunstomized model' requirements. The modification of the file is shown in the figure below:
[image]
Then I execute the command "python quant.py --model_file_path checkpoint_iter_370000.pth --quant_mode float --inspect", and encountered an error:
```
No CUDA runtime is found, using CUDA_HOME='/usr/local/cuda'

[VAIQ_NOTE]: Loading NNDCT kernels...
-------- Start openpose_lightweight test 

[VAIQ_NOTE]: =>Inspector is initialized successfully with target:
name: DPUCZDX8G_ISA1_B4096
type: DPUCZDX8G
isa_version: 1

[VAIQ_NOTE]: =>Start to inspect model...

[VAIQ_NOTE]: =>Quant Module is in 'cpu'.

[VAIQ_NOTE]: =>Parsing PoseEstimationWithMobileNet...

[VAIQ_NOTE]: Start to trace model...

[VAIQ_NOTE]: Finish tracing.

[VAIQ_NOTE]: Processing ops...
██████████████████████████████████████████████████| 152/152 [00:00<00:00, 760.48it/s, OpInfo: name = return_0, type = Return]                                                                              

[VAIQ_NOTE]: =>Doing weights equalization...
/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/nndct_shared/optimization/commander.py:454: RuntimeWarning: invalid value encountered in true_divide
  scale = np.where(sqrt_of_ranges != 0, range_0 / sqrt_of_ranges, scale)
/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/nndct_shared/optimization/commander.py:454: RuntimeWarning: divide by zero encountered in true_divide
  scale = np.where(sqrt_of_ranges != 0, range_0 / sqrt_of_ranges, scale)

[VAIQ_NOTE]: =>Quantizable module is generated.(quantize_result/PoseEstimationWithMobileNet.py)
Traceback (most recent call last):
  File "quant.py", line 307, in <module>
    model_name=model_name,
  File "quant.py", line 229, in quantization
    # start to inspect
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/apis.py", line 182, in inspect
    self._inspector_impl.inspect(module, input_args, device, output_dir, verbose_level)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/hardware/inspector.py", line 84, in inspect
    self._partition.simple_allocate_op_device(dev_graph) 
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/hardware/dpu_partition.py", line 106, in simple_allocate_op_device
    ret, filter_msg = filters[node.op.type](node, self._dpu_target)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/hardware/dpu_op_filter.py", line 461, in filter_depthwise_conv2d
    ret, msg = check_load_jump_write(ic, channel_parallel, dilation)
UnboundLocalError: local variable 'channel_parallel' referenced before assignment

```
How can I fix it?
Thx!

mehdi-nait, 2022-07-06:
I'm currently having the same issue, it seems that something is wrong with the check_load_jump_write function. I'm currently looking into it, but if someone has a fix I'd really appreciate it !

carrotandjzy, 2022-07-11:
Hi @Truthonlyone 
This bug can be fixed by adding one line in line 416:
 channel_parallel = alu_engine .channel_parallel

carrotandjzy, 2022-07-11:
Hi @mehdi-nait 
Currently, the inspector is only an experimental feature. It does not have a complete check passes in the compiler. Therefore, in some cases, the results may be inconsistent. It would be great if you could share your case.

Truthonlyone, 2022-07-12:
> Hi @Truthonlyone This bug can be fixed by adding one line in line 416: channel_parallel = alu_engine .channel_parallel

Thanks!

closed 2022-07-12 by Truthonlyone