Xilinx/Vitis-AI #909: Fingerprint problem during xmodel compilation

alkalir, 2022-07-06:
Hello, 
I have a Versal hardware platform implemented on VCK190, with Petalinux running on top of Dual-core A72, plus AI engine connected, and I am trying to compile a custom neural network with Vitis AI 2.0. However, there is a mismatch between the DPU fingerprint I have on hardware, and the model fingerprint. More specifically, the DPU fingerprint is the following:
0x6000000f6088831 (DPU Arch: DPUCVDX8G_ISA0_B8192C32B3_DW )
while the model fingerprint is 0x603000b16011861.
I tried to explicitly set correct DPU Arch inside the json file, but I obtain the following error:
[image]

How can I solve this problem without regenerating a new hardware? 
Thank you, Giacomo

qianglin-xlnx, 2022-07-07:
Hi @alkalir 
For the DPUCVDX8G_ISA0_B8192C32B3_DW arch, you can use the Docker1.3.1 to build the model.

alkalir, 2022-07-08:
Hi @qianglin-xlnx , thank you for your answer. I tried to pull the Docker 1.3.1 version but when going to quantize the model I obtain the error with a log attached to this message. 
[1_quantizationAge_tf2.log](https://github.com/Xilinx/Vitis-AI/files/9065001/1_quantizationAge_tf2.log)
I report the source code used for quantization in the following:

```
model = keras.models.load_model("build/weights/AgeGen/Age/age_model.h5")
df = pd.read_csv('build/dataset/morph/morph_2008_nonCommercial.csv')

datagen = ImageDataGenerator(
    validation_split=0.999,
)

df = df.sample(frac=1, random_state=SEED).reset_index(drop=True)
split_row_idx = int(len(df) * 0.2)
df_train = df.iloc[split_row_idx:, :]

calibration = datagen.flow_from_dataframe(
	    dataframe=df_train,
	    directory='build/dataset/morph/',
	    x_col='photo',
	    y_col='age',
            batch_size = 1,
	    class_mode='raw',
	    target_size=(240,200),
	    subset='training',
	    seed=SEED,
	    save_to_dir='debug')
	   
	   
	  
quantizer = vitis_quantize.VitisQuantizer(model)

quantized_model = quantizer.quantize_model(
       calib_dataset = calibration, 
       #calib_batch_size = 1,
       #train_with_bn = False,
       #freeze_bn_delay = -1,
       #replace_sigmoid = False,
       fold_conv_bn = True,
       fold_bn = True,
       replace_relu6 = True,
       include_cle = True,
       cle_steps = 10)
       #,
       #forced_cle = True)
       
      
quantized_model.save('build/quantized_results/AgeGen/Age/quantized_model.h5')

```

This never happened starting from Vitis AI 1.4. Is it possible to install the target available in Docker in 1.3.1 inside the Docker 1.4?

qianglin-xlnx, 2022-07-12:
Hi @alkalir 
DPUCVDX8G_ISA0_B8192C32B3_DW is not supported in Vitis AI1.4 and later. So you can't install the target available in Docker in 1.3.1 inside the Docker 1.4.
Maybe you can use Docker1.4 to do the model quantize, then use docker1.3.1 to compile the model for DPUCVDX8G_ISA0_B8192C32B3_DW arch.

alkalir, 2022-07-12:
Hi @qianglin-xlnx , I tried to quantize using Docker 1.4 and compile using Docker 1.3.1, but there are still errors of not supported keras activation (error log is attacked
[image]

Why is a problem adding an old configuration file for target_library to a new version of Vitis-AI? Is that a hardware problem? Can we fix it by recompiling the target_factory library in VART?
The other solutions would be to create a different neural network, using only supported functions, or to recreate the hardware with a different DPU. Both solutions are very time consuming and I am not sure they will converge to a stable and working neural network accelerated by the AI Engine. 

Thank you for your support.

zhenzhen-AMD, 2022-07-13:
Hi @alkalir ，
Please provide the floating point model and the quantized model and the code for the quantized model. thanks.

qianglin-xlnx, 2023-06-25:
Hi @alkalir 
Please try VAI2.5 or VAI3.0. Our engineer verify the model passed with VAI2.5 and VAI3.0.