Xilinx/Vitis-AI #910: Runner of Xmodel Reporting Signed Input, But Was calibrated With Unsigned Values

Hamptonjc, 2022-07-09:
The input to my model in PyTorch are float images with pixel values scaled to the range of [0,1]. However, the xmodel is reporting signed inputs.

During calibration, I placed an assert statement to ensure no pixel value is below 0.0. The assert doesn't stop the calibration, thus I'm sure all pixel values are in the range [0,1].

Anyway to fix this? Thanks!

niuxjxlnx, 2022-07-27:
@Hamptonjc:

What's the meaning of the "signed inputs" you mentioned? As I know, value in range [0, 1] can be expressed by both signed and unsigned variable.

Hamptonjc, 2022-07-27:
@niuxjxlnx Correct, however given that my float input pixels are in the range [0,1] (i.e. not negative), when expressing them in signed fixpoint notation I am losing _precision_

For example: The Vitis-AI quantizer recorded that the fixed position was at 6. This means the smallest fraction I could represent is 1/64. If using unsigned, Vitis-AI _should_ record that the fixed position is at 7, thus the smallest fraction I could represent would be 1/128

niuxjxlnx, 2022-08-11:
@Hamptonjc :

Vitis-AI deafult setting is signed quantization, but you can config to have unsigned quantization. Please refer to this part:
https://github.com/Xilinx/Vitis-AI/tree/master/src/Vitis-AI-Quantizer/vai_q_pytorch#configuration-of-quantization-stategy