Xilinx/Vitis-AI #911: A questions about LeakyReLU is assigned to the CPU (Pythorch framework)

Truthonlyone, 2022-07-10:
Ubuntu version: 18.04
Vitis AI version: 2.5
Framework: Pytorch
DPU name: DPUCZDX8G_ISA1_B4096

The code of network about LeakyReLU:
```
def conv_dw_no_bn(in_channels, out_channels, kernel_size=3, padding=1, stride=1, dilation=1):
    return nn.Sequential(
        nn.Conv2d(in_channels, in_channels, kernel_size, stride, padding, dilation=dilation, groups=in_channels, bias=False),
        nn.LeakyReLU(0.1, inplace=True),

        nn.Conv2d(in_channels, out_channels, 1, 1, 0, bias=False),
        nn.LeakyReLU(0.1, inplace=True),
    )
```

I inspected float models before quantization, and the inspection results is:
[image]

Then, I quantized the model, and got the .xmodel successfully.

However, the following message is displayed after I executed the vai_c_xir command:
[image]

And I put the compiled model into Netron to check the LeakyReLU part, it looks like this:
[image]

It seems that conv2d+LeakyReLU can be depolyed to the DPU, while depthwise-conv2d+LeakyReLU cannot???
How can I solve it?

Truthonlyone, 2022-07-12:
> Hi @Truthonlyone You can refer to #593 For LeakyRelu, the alpha should be 0.1015625

Thanks for your suggestion.

I check the value of alpha using the Netron, and find that it has been changed to 0.1015625 although it is set to 0.1 in my code. May be the quantier do this change automatically for the users.

wanghong4compiler, 2022-07-12:
Hi @Truthonlyone ,
Yes, you are correct. Currently, DPU only supports conv2d+LeakyRelu.
For depthwise-conv2d, the activation can only be Relu or Relu6.

JasonXijieJia, 2022-07-13:
Hi @Truthonlyone,
  Thanks for your attention.
  To enable depthwizeconv2d + leakyrelu running on dpu, please customize your configuration referring to 
- [pg338/ReLU-Type for vivado flow](https://docs.xilinx.com/r/en-US/pg338-dpu/ReLU-Type): set ALU ReLU Type  as RELU+LEAKYRELU+RELU6
- [pg338/ReLU-Type for vitis flow](https://docs.xilinx.com/r/en-US/pg338-dpu/Customizing-and-Generating-the-Core-in-the-Vitis-IDE):  `define ALU_RELU_LEAKYRELU_RELU6

Then rebuild project and recompile the model with new generated arch.json file

qianglin-xlnx, 2022-07-27:
Hi @Truthonlyone 
Is this issue solved?

Truthonlyone, 2022-07-27:
> Hi @Truthonlyone Is this issue solved?

I replace the LeakyReLU with ReLU and retrain the modle to bypass this problem.

closed 2022-07-29 by qianglin-xlnx