Xilinx/Vitis-AI #913: How do I use my model

easilee, 2022-07-11:
I have my own trained model and want to complete do inference on ZCU102

qianglin-xlnx, 2022-07-11:
Hi @easilee 
You can refer to https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Quick-Start to deploy your model on ZCU102.

easilee, 2022-07-11:
[image]

easilee, 2022-07-11:
[image]

easilee, 2022-07-11:
it shows nan

easilee, 2022-07-11:
[image]
this is xmodel input

tracksnow, 2022-07-12:
From your above comments, this has nothing to do with xmodel, it's only the basic python program knowledge. Please remove the xmodel & vatis-ai part, only leave the basic python logic to debug your issue.  Thanks.

lishixlnx, 2022-07-12:
From your above comments, this has nothing to do with xmodel, it's only the basic python program knowledge. Please remove the xmodel & vatis-ai part, only leave the basic python logic to debug your issue. Thanks.

easilee, 2022-07-12:
> 从您上面的评论来看，这与xmodel无关，只是基本的python程序知识。请删除 xmodel & vatis-ai 部分，只留下基本的 python 逻辑来调试您的问题。谢谢。

https://github.com/Xilinx/Vitis-AI-Tutorials/tree/1.4/Feature_Tutorials/tf2_quant_fine_tune/files/application

I use this

lishixlnx, 2022-07-13:
that is different. the sample is "dtype=np.int8" while you use float32.
please follow the guide carefully. thanks.

qianglin-xlnx, 2022-08-09:
Hi @easilee 
Is this issue solved?

qianglin-xlnx, 2022-08-23:
Hi @easilee 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-08-23 by qianglin-xlnx