Xilinx/Vitis-AI #914: docker build fail on v2.5

lp6m, 2022-07-11:
I am trying Vitis-AI [v2.5](https://github.com/Xilinx/Vitis-AI/tree/v2.5).

```
git clone --recursive https://github.com/Xilinx/Vitis-AI.git -b v2.5
cd Vitis-AI/docker
./docker_build_gpu.sh
```

An error occurred creating `vitis-ai-pytorch`  conda environment. 
What should I do?

```
Step 53/61 : RUN if [[ ${VAI_CONDA_CHANNEL} =~ .*"tar.gz" ]]; then        cd /scratch/;        wget -O conda-channel.tar.gz --progress=dot:mega ${VAI_CONDA_CHANNEL};        tar -xzvf conda-channel.tar.gz;        export VAI_CONDA_CHANNEL=file:///scratch/conda-channel;     fi;     . $VAI_ROOT/conda/etc/profile.d/conda.sh     && mkdir $VAI_ROOT/conda/pkgs     && mamba install -c conda-forge conda-build     && conda config --env --remove-key channels     && python3 -m pip install --upgrade pip wheel setuptools     && conda config --env --append channels ${VAI_CONDA_CHANNEL}     && mamba env create -f /scratch/vitis-ai-optimizer_darknet.yml     && mamba env create -f /scratch/vitis-ai-optimizer_pytorch.yml     && mamba env create -f /scratch/vitis-ai-optimizer_tensorflow.yml     && mamba env create -f /scratch/vitis-ai-optimizer_tensorflow2.yml     && mamba env create -f /scratch/vitis-ai-pytorch.yml         && conda activate vitis-ai-pytorch         && pip install graphviz==0.19.1     && mamba env create -f /scratch/vitis-ai-tensorflow.yml         && conda activate vitis-ai-tensorflow         && pip install -r /scratch/pip_requirements.txt     && mamba env create -f /scratch/vitis-ai-tensorflow2.yml         && conda activate vitis-ai-tensorflow2         && mamba install --no-update-deps -y vai_q_tensorflow2_gpu pydot pyyaml jupyter ipywidgets             dill progressbar2 pytest scikit-learn pandas matplotlib             pillow -c conda-forge -c defaults         && pip install -r /scratch/pip_requirements.txt         && pip install --ignore-installed tensorflow==2.8 keras==2.8 protobuf==3.11.*         && pip uninstall -y h5py         && mamba install -y --override-channels --force-reinstall h5py=2.10.0 -c conda-forge -c anaconda     && conda clean -y --force-pkgs-dirs     && rm -fr ~/.cache     && mkdir -p $VAI_ROOT/conda/pkgs     && sudo chmod 777 $VAI_ROOT/conda/pkgs     && sudo rm /opt/vitis_ai/conda/.condarc     && mkdir -p $VAI_ROOT/compiler         && conda activate vitis-ai-pytorch         && sudo cp -r $CONDA_PREFIX/lib/python3.7/site-packages/vaic/arch $VAI_ROOT/compiler/arch     && cat /dev/null > /etc/conda/condarc     && cat /dev/null > ~/.condarc
 ---> Running in fe5f1f33f803

python: /home/conda/feedstock_root/build_artifacts/libsolv_1625855492383/work/src/rules.c:261: solver_addrule: Assertion `!p2 && d > 0' failed.
/bin/bash: line 1:   323 Aborted                 (core dumped) mamba env create -f /scratch/vitis-ai-pytorch.yml
```

janifer112x, 2022-07-12:
Hi ，
I tried with your command, it will get the   Vitis-AI V2.5 tag, which was on June 15.
we've fixed the problem in the master branch, Could you please try with the latest code in master branch  and feel free to let us know if you still have problem?

lp6m, 2022-07-12:
Thank you for your reply.
I confirmed that docker build success on latest master branch.