Xilinx/Vitis-AI #915: Can not run quantization with Hardswish operator on v2.5

SuHuynh, 2022-07-12:
I created simple model to test quantization with hardswish, like this:

```
class Test_Model(nn.Module):
	def __init__(self):
		super().__init__()
		self.conv1 = nn.Conv2d(3, 32, 3)
		self.hard_swish = nn.Hardswish()
	def forward(self, x):
		x1 = self.conv1(x)
		x2 = self.hard_swish(x1)
		return x2
```

I got this error when run quantization- mode calib
[image]

It seams a bug in pytorch_nndct/nn/modules/hardswish.py. So please fix it. Thanks

niuxjxlnx, 2022-07-18:
@SuHuynh:

This is really a bug. You can remove that line pytorch_nndct/nn/modules/hardswish.py:40 for work around. 
The fix will be in next release pakage.

closed 2022-07-18 by niuxjxlnx

lino-alves, 2022-07-18:
I had the same issue and commented line 40 of pytorch_nndct/nn/modules/hardswish.py, but then there are two arguments missing in line 53:
`output = fake_quantize_per_tensor(output, scale_inv=128, zero_point=0, quant_min=-128, quant_max=127)`
**_method_** and **_inplace_** arguments are required. What should be passed here?

This seems to be a bug too and I expect the OP will have the same problem.

niuxjxlnx, 2022-07-19:
@lino-alves :

You can set method=2, inplace=self.quantizer.inplace